qmk/qmk_firmware #1240: Layer indicator using ws2812 led strip

bartiosze, 2017-04-18:
Hi,
I'm looking for inspiration or pointers where to start if I would like to add a layer indicator using ws2812 led strip.  Sometimes my keyboard (atreus) gets stuck on a layer and I have problems switching to a default one. I would like to add either a dedicated diode and assign colors to layers or use a led strip for that.

Thanks in advance for any suggestions.
Regards,
ba

priyadi, 2017-04-18:
My handwired keyboard uses RGB LED strip for layer indicator, among other things. Those are SK6812 minis actually, but electrically should be compatible with WS2812s.

https://github.com/qmk/qmk_firmware/tree/master/keyboards/handwired/promethium/keymaps/priyadi

bartiosze, 2017-04-28:
@priyadi your build looks amazing. 
I will try to steal some ideas from your LED indicators. Those look dope!
I hope I will be able to figure out the code;)

closed 2017-06-24 by skullydazed