qmk/qmk_firmware #1241: Pro Micro not Resetting (I know this isn't directly related, but just looking for help)

whitef0x0, 2017-04-18:
Hi all,

I recently had the unfortunate experience of somehow bricking my Arduino Pro Micro. After searching online for a whole day, I still haven't narrowed down how to fix my board.

Currently the main indicator light comes on when plugged in but I can't flash to the device as it won't reset (and I've checked my reset button and it is fully functional) and it is not being recognized on any of my `/dev/tty.*` or `/dev/cu.*` ports.

Anyone have any advice? I've used USBProber on Mac and looked at my Console output but haven't been able to see any USB connection errors.

This happened after I pulled out my usb cord rather aggressively, but no damage is visible on the device anywhere.

Any advice would be much appreciated, if possible!

whitef0x0, 2017-04-18:
Also, forgot to add, while the Pro Micro is flashed with a working keymap, it is not functioning at all/outputting any characters while typing.

fredizzimo, 2017-04-19:
The first thing I would try is another USB cable. If that doesn't help, then re-solder the USB port on the keyboard.

drelephant, 2017-04-19:
Have a look [here](https://arduino.stackexchange.com/questions/6702/arduino-pro-micro-clone-not-recognized-flickering-rx-led). I've had similar problems with my Pro Micros before, it's irritating. Usually I just keep trying the double reset thing and hitting upload in the IDE. You need to be fast. It's because a bad sketch can break the usb connection.

Also [this link](https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide#troubleshooting-and-faq) for the sparkfun troubleshooting page.

whitef0x0, 2017-04-19:
@drelephant I'll try another cable. It's not in a reset loop though, just light on with no flickering and no USB connection.

skullydazed, 2017-06-24:
I'm closing this to clean up our open issues, but if you still have questions @whitef0x0 please reopen this or open a new issue.

closed 2017-06-24 by skullydazed

BenRoe, 2017-07-31:
I wrote a tutorial how to use a pro micro as an keyboard controller. At the end you find some hints how to flash the pro micro on a mac or linux.
https://maker-tutorials.com/arduino-pro-micro-als-keyboardtastatur-controller-teensy-2-0-alternative/