qmk/qmk_firmware #1247: OSM delay?

lisinge, 2017-04-19:
I'm interested in using `OSM(MOD_LSFT)` as my main LeftShift.
The problem i'm having is that i play a lot of games where you use LeftShift (CS:GO) for example, and i feel like there is a a slight delay when holding it down which makes it unusable while gaming.

One solution would be to have a game layer where i just use `KC_LSFT`, but i really don't want to have to switch to a layer while gaming.

I've been trying to find the code that handles `OSM()` but i'm afraid i'm just too new to this codebase.

Is the delay configurable or do you have another solution that would fit? And could someone point me in the right direction to understand how OSM works?

fredizzimo, 2017-05-12:
I didn't see anything obvious in the code that would cause a delay, but I didn't look too carefully.

I suggest that you make a dedicated gaming layer with normal shift behaviour.

mneme, 2017-06-19:
I have the same issue but not a problem for gaming. OSM+shift does give a noticeable delay before activating the OSM state.

Maartenwut, 2017-06-19:
Yes, there is a delay. I can notice it while playing Borderlands 2 or other games. Like @fredizzimo said, the best solution is to make one layer for games.

lisinge, 2017-06-19:
The best solution IMO would be if the behaviour would be that when you press the key with `OSM(MOD_LSFT)` then `LSFT` gets sent instantly and if let go fast it acts as one shot but if held for longer then it acts as regular `LSFT`.

The current behaviour feels like the timer is used to check if the key is actually held down or just quickly pressed and after it has decided it finally sends `LSFT`.

But maybe this can be accomplished with something other than `OSM`?

drashna, 2018-03-25:
Personally, I have a layer that reverts OSM's to their normal keys. 
That and a gaming layer.

micke, 2018-10-05:
This is no longer an issue.

closed 2018-10-05 by micke