qmk/qmk_firmware #1248: Cant make tap dance work.

YamiFrankc, 2017-04-20:
Hello. sorry for posting so much, but im not a coder and its harder for me to solve problems just looking at the code.

I have added "tap_dance_enable = yes" to qmk/keyboards/ergodox/infinity/Makefile , and "#define TAPPING_TERM 200" to qmk/keyboards/ergodox/infinity/config.h

And the output of compiling is: 
Linking: .build/ergodox_infinity_YamiFrankc.elf                                               [ERRORS]                                                                                                                                .build/obj_ergodox_infinity_YamiFrankc/quantum/process_keycode/process_ucis.o: In function `process_ucis':            | /mnt/d/QMK/qmk_firmware/quantum/process_keycode/process_ucis.c:99: undefined reference to `ucis_symbol_table'         | collect2: error: ld returned 1 exit status 

Then, I copied the keymap.c and Makefile from algernon into my custom Keymap directory, and it compiles directly.

Then, I kept the Makefile and copied the simplest tap dance definition from him and added it into my own keymap. and it still wont work.

I was trying to eliminate errors mixing his files with mine, so I think that there is something wrong in my keymap.c(which is based in the default ergodox one), but cant quite figure out that is wrong, here is it:

[Pastebin](https://pastebin.com/MWD2Pc16)

ryanmaclean, 2017-04-30:
Are you possibly missing this part of @algernon's `keymap.c`?

```
void qk_ucis_symbol_fallback (void) {
  for (uint8_t i = 0; i < qk_ucis_state.count - 1; i++) {
    uint8_t code;

    if ((qk_ucis_state.codes[i] >= M(A_1)) && (qk_ucis_state.codes[i] <= M(A_0)))
      code = qk_ucis_state.codes[i] - M(A_1) + KC_1;
    else
      code = qk_ucis_state.codes[i];
    register_code(code);
    unregister_code(code);
    wait_ms (10);
  }
```

algernon, 2017-05-01:
I think the problem may be in your `Makefile`. Does it have an `UCIS_ENABLE=yes` line? If so, remove that. The compilation fails due to a missing UCIS symbol table, but that is not required for tap-dance.

closed 2017-06-24 by skullydazed