qmk/qmk_firmware #1250: Windows install broken?

codepoke-kk, 2017-04-21:
I'm a rookie and attempting to follow the instructions to get this project working on my computer for my ErgoDox. I've obviously missed something, but can't tell what. 

I’ve downloaded and installed MHV AVR Tools, MinGW, and QMK zip. I’ve followed all the steps to the best of my skills on a Win 7 box, and ran the two setup batch scripts. The driver install failed, so I did that install directly using the included driver installation .exe and it reports success. When I attempt to run make from the command prompt, it reports failures of date.exe and dies, then fails to run cmp.exe probably trying to report on success/failure. So, it sounds like I have make in my path, but several things are missing. I’ll try to figure out what they are at my next opportunity.

I ran Win_Check.bat and will attach the output

[Win_Check_Output.txt](https://github.com/qmk/qmk_firmware/files/948348/Win_Check_Output.txt)

codepoke-kk, 2017-04-23:
I figured it might be conflicts with other installs, so I installed it on another computer that's never flashed an Arduino or anything similar and got the same set of symptoms. I did the only thing I know to do, and ran a make in my ergodox folder while running ProcMon. There's obviously a lot of stuff going on. The first thing to go bad, though, is date.exe fails. 

Sh.exe finds date.exe in MHV AVR Tools\bin and calls it using a mix of SysWOW64 and System32 calls to wow654. Date.exe starts and finds a number of files in c:\MinGW\msys\1.0\bin. It's last action is to find sechost.dll. Immediately thereafter a svchost gets information on date.exe and WerFault.exe is called to create a crash report. The svchost is secsvcs and the path says WindowsDefender\Scans\FailTelemetry. 

This all seems promising.

codepoke-kk, 2017-04-23:
Promising is not a promise. With both Windows Defender and Kaspersky AV disabled (and not obviously causing problems visible in ProcMon), the symptoms are unchanged. The data.exe stack dump is hex to me:
MSYS-1.0.18 Build:2012-11-21 22:34
Exception: STATUS_ACCESS_VIOLATION at eip=60897244
eax=000001F4 ebx=60EA022C ecx=0000000F edx=00000000 esi=000001F4 edi=60EA022C
ebp=0028FEA8 esp=0028FE90 program=c:\Program Files (x86)\MHV AVR Tools\bin\date.exe
cs=0023 ds=002B es=002B fs=0053 gs=002B ss=002B
Stack trace:
Frame     Function  Args
0028FEA8  60897244  (60EA022C, 000001F4, 0028FEFC, 60EA2704)
0028FEC8  60808A3F  (60EA00D4, 000001F4, 00000050, 0028FEFC)
0028FF18  60804426  (6089C740, FFFFFFFE, 00000050, 6089C6A4)
0028FF58  60804995  (00000000, 00000000, 00000000, 00000000)
0028FF78  0040A75F  (004014C0, 037F0000, 0028FF94, 7710336A)
0028FF88  0040103E  (7EFDE000, 0028FFD4, 779A9902, 7EFDE000)
0028FF94  7710336A  (7EFDE000, 777FAC6C, 00000000, 00000000)
0028FFD4  779A9902  (00401000, 7EFDE000, 00000000, 00000000)
0028FFEC  779A98D5  (00401000, 7EFDE000, 00000000, 78746341)
End of stack trace

fredizzimo, 2017-04-23:
This is the same symptom that has been discussed a few times before. The `msys` package is missing. So you need to install it from the mingw command prompt with `mingw-get install msys`. I guess no-one has got around to update the `2-setup-environment-win.bat` to include it yet. 

If this fixes it for you I will update the bat file.

codepoke-kk, 2017-04-23:
I appreciate the thought, fredizzimo, but running `c:\MinGW\bin>.\mingw-get.exe install msys` ran successfully without changing my symptoms at all. The make command resulted in the same stack traces.

drashna, 2018-03-25:
This should be fixed now.  #448, as well as using MSYS2.

closed 2018-03-25 by drashna