qmk/qmk_firmware #1251: I can't get the two sides of lets_split to talk to each other

geoffreyborggaard, 2017-04-21:
Both sides work fine.  I've got pins 2 and 3 connected and pulled up to VCC (10K resistors on one side 4.7K on the other).

I've flashed the rev2_i2c rom on to them.  Customized slightly for my layout.

Both sides work fine when plugged in, but they don't appear to be able to communicate when only one side is plugged in.   I'm using the EE_HANDS approach (verified working) to determine which side is which.

Is there any way to get some debugging info as to where it might be failing?

Justinicus, 2017-04-24:
When you say you have 10k resistors on one side, and 4.7k on the other, do you mean you put resistors on both PCBs?

geoffreyborggaard, 2017-04-24:
I'm using two Arduino Micros with hand wired switches.  No PCBs.

geoffreyborggaard, 2017-05-23:
I was able to hook it up to a scope.  When I put the scope in I2C decode mode, I see that its just constantly sending the same thing.  The scope decodes it as "W: 19 ?"  I'm not sure what that means.

geoffreyborggaard, 2017-05-23:
Another oddity.  The scope says its running at 227 khz, but i2c.h has:
#define SCL_CLOCK  400000L

I wonder if the clock speed on the Arduino Micro is wrong?

nooges, 2017-05-25:
What if you try setting SCL_CLOCK down to 100000L, that's what that value used to be before someone else changed it. If you want to stick with 400k, then it's better to drop the 4.7k resistors to 1.8k.

geoffreyborggaard, 2017-05-31:
I tried that.  It didn't work, but I haven't put it on the scope yet to see what it looks like.

Question:   If I want to try "serial" instead, do I need to remove the resistors?

nooges, 2017-06-04:
Yeah, remove the resistors.

geoffreyborggaard, 2017-06-13:
I was able to set it up to a scope after dropping SCL_CLOCK as suggested.

I see the start condition, and then the address for the slave (0x32) being written.  It appears to be followed by a 1 and 0 (binary) and then both lines go high.   I don't see any ACK from the slave.

I tried removing the resistors and just going with serial.   No dice there either.  The scope just shows the line being pulled low for a very short tick at regular intervals.

drashna, 2018-03-25:
Were you able to solve this?

geoffreyborggaard, 2018-03-25:
I did a bunch of things.   I gave up on let's split and used the firmware that it was based on. (I forget the name). But that suffered the same problem.  Eventually I dropped the clock speed of the bus by 4x or so and it started working.

geoffreyborggaard, 2018-03-25:
Oh. I think I went with serial instead of i2c.  Pretty sure serial at the low bus speed was what worked.

drashna, 2018-03-25:
Well, I'm glad to hear that you were able to get it working! 

And if this is the 3.3v/8MHz ProMicro controller, I think there is a hack to fix it.  
Either way, that's pretty weird! 

However, I am going to close this.  If needed, reopen it or open a new issue.  
Thanks!

closed 2018-03-25 by drashna