qmk/qmk_firmware #1254: Adding support for STM32F04

skullydazed, 2017-04-23:
I'm attempting to use QMK for a new board I've designed: https://github.com/qmk/qmk_firmware/tree/simontester/keyboards/simontester

I can build and flash the firmware, but it never shows up on USB. I have a .bin file that flashes one of my LED's, and I'm able to successfully flash and use that. I have another .bin that initializes as a usb joystick, and I'm able to flash over USB, so I know the hardware for this board is fine. 

The chip is an STM32F042G6U6 and I've created `boards/GENERIC_STM32_F04` to support it. Most of what I took is cribbed from ST_NUCLEO32_F042K6 and other STM32 boards. I've tried everything I know but unfortunately I did not leave the SWD headers exposed so I'm a little stuck as to what to check from here.

Anyone have an idea what to poke at next?

fredizzimo, 2017-04-23:
I don't know what could be wrong, but I suggest that you start removing stuff from `https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/chibios/main.c` and see if that helps. 

I think you can reduce it to something like this 

```c
/* ChibiOS/RT init */
halInit();
chSysInit();
/* Init USB */
init_usb_driver(&USB_DRIVER);
while(true) {
    chThdSleepMilliseconds(50);
}
```

The USB should still initialize and show up with that code. But obviously there's no keyboard loop or anything so it won't work as a keyboard.

skullydazed, 2017-04-23:
Even stripping down to this leaves me in a non-working state:

```
/* Main thread
 */
int main(void) {
  /* ChibiOS/RT init */
  halInit();
  chSysInit();

  // TESTING
  // chThdCreateStatic(waBlinkerThread, sizeof(waBlinkerThread), NORMALPRIO, blinkerThread, NULL);

  /* Init USB */
  init_usb_driver(&USB_DRIVER);

  /* Main loop */
  while(true) {
    chThdSleepMilliseconds(50);
  }
}
```

skullydazed, 2017-04-23:
I've figured out what this board needs to work:

```
        /* Remap PA11-12 to PA9-10 for USB */
        RCC->APB2ENR |= RCC_APB2ENR_SYSCFGCOMPEN;
        SYSCFG->CFGR1 |= SYSCFG_CFGR1_PA11_PA12_RMP;
```

If I place that immediately after `init_usb_driver(&USB_DRIVER);` everything works fine. Now I just need to figure out the best way to inject that code into the USB init routines.

fredizzimo, 2017-04-23:
Nice. But shouldn't the pin remapping be done before the USB driver init, and not after it?

I think the most logical place would be to move that piece of code to `usb_main.c`, and add an ifdef to check for the chip. There is some other chip specific code in that file, so I don't think it's worth trying to generalize too much and move out to the keyboard init.

skullydazed, 2017-04-23:
I'm not sure which is best to be honest. The example code I was given has it after the USB initialization.  It seems to work either at the top or bottom of `init_usb_driver()`.

This is what I put in as a placeholder. Based on what you wrote it sounds like this may be the best way anyway:

```
$ git diff usb_main.c
diff --git a/tmk_core/protocol/chibios/usb_main.c b/tmk_core/protocol/chibios/usb_main.c
index d0c72c46..79f4b5c1 100644
--- a/tmk_core/protocol/chibios/usb_main.c
+++ b/tmk_core/protocol/chibios/usb_main.c
@@ -1026,6 +1026,16 @@ void init_usb_driver(USBDriver *usbp) {
   obqObjectInit(&console_buf_queue, console_queue_buffer, CONSOLE_EPSIZE, CONSOLE_QUEUE_CAPACITY, console_queue_onotify, (void*)usbp);
   chVTObjectInit(&console_flush_timer);
 #endif
+#ifdef STM32_REMAP_PINS
+ /* Remap PA11->PA9 and PA12->PA10 on STM32 devices that require it.
+  *
+  * The following MCU's are known to require this:
+  *
+  *     STM32F042x6
+  */
+  RCC->APB2ENR |= RCC_APB2ENR_SYSCFGCOMPEN;
+  SYSCFG->CFGR1 |= SYSCFG_CFGR1_PA11_PA12_RMP;
+#endif
 }

```

fredizzimo, 2017-04-23:
Actually, when I think about it,  the code belongs in the board file. I'm not sure if `early_init` or `board_init` is better. But both are probably run before the USB initialization, so they should be fine.

skullydazed, 2017-04-23:
How's this look? It seems to work for me.

https://github.com/qmk/qmk_firmware/commit/d89671d9871df49633281a5ba8df7e1196a4199f

fredizzimo, 2017-04-24:
Yes it looks good to me. At some point we should probably move all those board and drivers to some global place so that they can be shared between keyboards.

skullydazed, 2017-04-25:
Thanks for your help @fredizzimo. I agree that sharing board files in the future will be a good thing, I already have two other boards based around the F303 I'm working on that will use the same board file.

closed 2017-04-25 by skullydazed