qmk/qmk_firmware #1260: KC_UNDO, KC_CUT, KC_COPY, KC_PASTE keycodes stop working.

stratact, 2017-04-27:
I recently updated to HEAD a couple of days and rebuild/flash my keymap for my Ergodox EZ and these keys stop working.

fredizzimo, 2017-04-27:
A little bit more information wold be helpful.

* Could you share a link to your keymap?
* Do you know the commit hash of the previous version you had before the update? If you did a merge it should be easy to find out, by looking at parent of the previous merge. If you used rebasing then you can checkout old version by finding it's hash using `git reflog`

stratact, 2017-04-28:
* https://gist.github.com/stratact/37364dd74b25f07f8ad5e860924b8bc1
* I unfortunately removed the directory and did another git clone to HEAD and I don't know the commit hash of the previous version that I used when those keycodes work, but I would estimate it's something around 6 months ago.

stratact, 2017-05-05:
I'm tried commits from up to earlier Feburary with the same result of the keys not working. I can't build my firmware past that because I get errors saying it's missing `cmp` and needs `diffutils` which I had installed already.

packman86, 2017-05-29:
I am having issues getting KC_COPY and KC_PASTE on Tap Dance to work. Maybe it is related to OPs issue?

`TD_COPY_PASTE`

`qk_tap_dance_action_t tap_dance_actions[] = {
  // Other declarations would go here, separated by commas, if you have them

  //Tap once for Esc, twice for Caps Lock
  [TD_ESC_CAPS]  = ACTION_TAP_DANCE_DOUBLE(KC_ESC, KC_CAPS),
  // Tap Left FN once for Copy, twice for Paste
  [TD_COPY_PASTE] = ACTION_TAP_DANCE_DOUBLE(KC_COPY, KC_PASTE),
};
`

In the keymap layer 
`TD(TD_COPY_PASTE),`

nikchi, 2017-05-31:
@packman86 Hey, just want to put this out here. It may be that windows does not recognize KC_COPY and those sorts of keycodes as what they're labeled as here.

https://geekhack.org/index.php?topic=41989.msg2277717#msg2277717

Here's a workaround that was proposed in the thread above.
```
[COPA]  = ACTION_TAP_DANCE_DOUBLE(LCTL(KC_C), LCTL(KC_V))
```

closed 2017-06-27 by skullydazed