qmk/qmk_firmware #1261: Mousekeys acceleration options

bennypowers, 2017-04-28:
I'd like a little more control over mousekeys. To wit: Instead of the default acceleration being highest and accel modifiers slowing it down, I'd like the default accel to be 1 and modifiers to change it to 0 or 2. Alternately, I'd like to start at 0 and have modifiers change it to 1 or 2.

fredizzimo, 2017-05-12:
I think you are waiting for this pull request https://github.com/qmk/qmk_firmware/pull/1173.

You need to wait for @BrentBrightling or someone else to fix the issues first though. Or fix them yourself.

drashna, 2018-03-25:
Not sure if this helps or not.... but....
https://docs.qmk.fm/feature_mouse_keys.html#configuring-the-behavior-of-mousekeys

drashna, 2018-10-21:
I think this has been added since. 

If not, please reopen.

closed 2018-10-21 by drashna