qmk/qmk_firmware #1265: The Travis build is broken

fredizzimo, 2017-04-29:
It seems like the continous integration system didn't properly build https://github.com/qmk/qmk_firmware/pull/1231 - S60-X RGB support. As you can see in the Travis report [here](https://travis-ci.org/qmk/qmk_firmware/builds/225709278), it was only building the unit tests and not the actual code.

This causes my pull request https://github.com/qmk/qmk_firmware/pull/1264 to [fail](https://travis-ci.org/qmk/qmk_firmware/builds/227225682?utm_source=github_status&utm_medium=notification), because the above pull request don't build cleanly, at least I think it's that one, but it could also be some other. I strongly doubt the cause is my pull request since it doesn't touch anything even remotely related.

jackhumbert, 2017-04-30:
Ah, yeah - thanks for catching that. It looks like it's being caused by the `-` instead of the `_`?

jackhumbert, 2017-04-30:
Fixed it! Thanks :)

closed 2017-04-30 by jackhumbert