qmk/qmk_firmware #1271: RESET keycode for HHKB alt controller now working?

tengg, 2017-05-03:
I was using clueboard, which uses RESET kc to configure the keyboard into DFU mode.
But when I use the HHKB alt controller from hasu, this does nothing.

Is it just not work that way or I need to configure something else?

tengg, 2017-05-03:
I tried magic+pause, the hid_listen showed "Jumping to bootloader... [1]    8258 segmentation fault  ~/Downloads/hid_listen.mac", but dfu-programmer doesn't recognize it as boot loader.

Also tried key that mapped to "RESET", hid_listen doesn't show "jumping to bootloader", but just segmentation fault.

However, if I push the hardware button, dfu-programmer can see it.

tengg, 2017-05-04:
Updated.
I think there's something wrong in QMK code in low level. But I'm not quite sure...

skullydazed, 2017-05-04:
Check the Makefile or rules.mk, is BOOTLOADER_SIZE the same in both?

tengg, 2017-05-04:
You are right.
Tmk is 4096 and qmk is 512.
Why is that? And should I change to 4096 in qmk/hhkb?

skullydazed, 2017-05-04:
512 is the size for teensy. Other boards use a bootloader that is 4096 bytes. You should change that to match in qmk/hhkb.

cinaeco, 2017-05-04:
I encountered the same issue before, but didn't think much about it. Updating `BOOTLOADER_SIZE` to 4096 does indeed allow the `RESET` keycode to work. We should change it! Unless anyone knows the reason it was set to 512?

tengg, 2017-05-04:
Thanks! It worked.
The comment in rules.mk is "As per original Hasu settings".  But in TMK it's 4096 now. I agree that we might want to change that too! Unless there's a particular reason.

skullydazed, 2017-05-04:
We'd love to have this fix if one of you wants to submit a PR. Otherwise it'll end up waiting until we have a chance to get around our own PR. :)

tengg, 2017-05-04:
I'll do the PR, and close this issue.

closed 2017-05-04 by tengg