qmk/qmk_firmware #1276: Tap Dancing with special keys (RESET, TG(), etc.) does not work

antelaurijssen, 2017-05-05:
HI everyone,
I'm trying to get tap dancing to work on my new S60X with some special keys, but I'm not having any luck. For example, i tried to get my Escape key to give KC_ESC when tapped once, and RESET when double tapped. That did not work. But if I try KC_ESC for one tap and KC_CAPS for a double tap, that works. Same thing when I tried assigning TG(1) to a double tap on my left control key. That didn't work. But KC_A on double tap of left control worked. Am I missing something or is that normal behaviour with these special keys?

Thanks!

nikchi, 2017-05-31:
This seems like normal behavior with QMK. It seems like the tap dance double macro just handles two regular keycodes.

Looks like RESET and TG(layer) are specially defined in quantum_keycode.h and are handled differently

Perhaps one solution would be to use ACTION_TAP_DANCE_FN_ADVANCED to switch layers instead?

antelaurijssen, 2017-05-31:
Hi nikchi. Yes, after digging into this a little more I came to the same conclusion. I'll look into ACTION_TAP_DANCE_FN_ADVANCED. Thanks!

closed 2017-05-31 by antelaurijssen