qmk/qmk_firmware #1281: Ergodox EZ: missing <inttypes.h>, can't compile

adiabatic, 2017-05-06:
It's been a while since I've changed my layout, but I wanted to change a few things. However, none of the Ergodox EZ layouts work anymore. I made sure to update the relevant homebrew utils:

```
brew tap osx-cross/avr
brew install avr-libc
brew install dfu-programmer
```

and I navigated to `qmk_firmware/keyboards/ergodox` and typed `make clean`. That worked fine, modulo one warning:

```
/bin/sh: arm-none-eabi-gcc: command not found
```

but when I try to build zweihander-osx, it errors out:

```
~/Projects/qmk_firmware/keyboards/ergodox $ make zweihander-osx
Making ergodox/ez with keymap zweihander-osx

avr-gcc (GCC) 6.3.0
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  18538	      0	  18538	   486a	ergodox_ez_zweihander-osx.hex

Compiling: keyboards/ergodox/ez/twimaster.c                                                        keyboards/ergodox/ez/twimaster.c:9:22: fatal error: inttypes.h: No such file or directory
 #include <inttypes.h>
                      ^
compilation terminated.
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [.build/obj_ergodox_ez_zweihander-osx/twimaster.o] Error 1
make: *** [zweihander-osx] Error 1
Make finished with errors
```

I get the same error if I run `make default`, and lots of the same error if I just run `make` (plus more that seem to be ergodox/infinity related). What should I do? I've already tried this both with my own up-to-date repo and also from a fresh .zip download and I get the same errors.

cquiroz, 2017-05-07:
I'm seeing the same error after merging the last code from upstream. I updated my tools with brew as well

fredizzimo, 2017-05-07:
If you do a search for `inttypes`, then this turns up.

https://github.com/qmk/qmk_firmware/issues/802

I bet the issue is still the same, brew seems to mess up.

cquiroz, 2017-05-08:
uninstalling and reinstalling the tools as is #802 solves the problem for me

adiabatic, 2017-05-08:
Uninstalling and reinstalling worked for me, too. Thanks!

closed 2017-05-08 by adiabatic