qmk/qmk_firmware #1285: About sleep led

Dbroqua, 2017-05-08:
Hello,

I want to make my own keyboard ( https://github.com/Dbroqua/MX_HHKB ) and I would like to know if it's possible to change the affected led to have the sleep mode instead.

In the code I've seen that the led used is the CAPS_LED. Is it possible to set another led/group of led by using var or by redefining the function for the keyboard directory?

Regards,
Damien

fredizzimo, 2017-05-12:
Not directly, but it would be easy to add support for it. The cleanest way is probably to do it the same ways as the `CUSTOM_MATRIX` configuration variable, and add `CUSTOM_SLEEP_LED`. If that's enabled, the `sleep_led.c` would not be compiled, and the keyboard would have to provide the implementation instead. You could probably copy the existing implementation and modify it with the correct pins.

Another alternative would be to declare the functions as weak, but I think it's probably better to have an all or nothing override here.

A third alternative would be to expose the pin numbers as defines. But that would not work if the keyboard has a different number of LEDs.

Dbroqua, 2017-05-13:
Hello,

Thank you for your reply. I'm not sure to understand all but I will try :D .

skullydazed, 2017-06-24:
I'm closing this to clean up our open issues, but if you still have questions @Dbroqua please reopen this or open a new issue.

closed 2017-06-24 by skullydazed

Dbroqua, 2017-06-24:
Ok,

No problem.

Regards,
Damien