qmk/qmk_firmware #1287: keyboard_report->mods cleared by tap dance?

sdothum, 2017-05-08:
As far as I can tell, keyboard_report->mods is not available to tap dance functions -- modifiers appear to be cleared when tap dance defined keycodes are pressed. Was hoping to be able to detect modifier states.

sdothum, 2017-05-09:
In lieu of finding out where in the tap dance (?) code the mods mask might be cleared, the work around I applied is simply to manage my own mods mask in the process_record_user section.

drashna, 2018-03-25:
I believe it removes them here:
https://github.com/qmk/qmk_firmware/blob/master/quantum/process_keycode/process_tap_dance.c#L105-L111

But yeah, your solution would work. 

Also, closing this.  If you need to, reopen it or open a new issue.

closed 2018-03-25 by drashna