qmk/qmk_firmware #1289: Timer during key press to delay layer switch not working?

skatardude10, 2017-05-11:
[Here's what I am talking about.](https://github.com/skatardude10/dotfiles/blob/6cc61dcc5bbfc149a87fddb39484a22b0d4f79c0/config/keymap.c#L335-L340)

**Background**: I would like to have my ACTION_LAYER_TAP_KEY be able to run code and switch layers when held and send KC_SPC when tapped, or similar functionality.

I having it working generally, and with timers running space gets entered just fine, but I am trying to set a timer at the beginning of the keystroke to wait to turn on the function layer only after a certain amount of time, otherwise I enter the other layers and enter space when tapped which causes me to miss keys when typing fast and not fully leaving the function layer. 

A timer as the key is held to only turn on the function layer after a certain amount of time has passed never seems to work. Do you guys have any ideas? 

In the link above I have highlighted the section that shows what I am trying to do, but doesn't work no matter how I try it... all sorts of while loops, if else statements, nested ifs... I have run out of ideas.

fredizzimo, 2017-05-12:
A loop like that will just stall the keyboard for 100ms, which is probably not what you want, and even if you wanted that, a sleep would be better.

What you should do instead is to add a `void matrix_scan_user(void)` function and check the timer there instead. You probably also need some other global variables for bookkeeping so that you know what to do on timeout.

skatardude10, 2017-05-12:
That is *exactly* what was happening.  

I thought about the matrix scan but hesitated to try it because of the warnings. I will try some conditions in there, mindful of being as light as possible, and post back if I get it working :-) 

Thanks for pointing me in this direction, I'll give it a try now!

skatardude10, 2017-05-13:
[Matrix scan](https://github.com/skatardude10/dotfiles/commit/cf408b5c96dd5e0d8d6f961c51d620946afae05e?diff=unified#diff-0b47e102997c5901cda7bfd33cdf69dc) worked great! 

Thanks again, now to consolidate stuff into functions and clean things up in my keymap.c, this issue is solved.

closed 2017-05-13 by skatardude10