qmk/qmk_firmware #1292: Using right side for connection with EE-hands on lets split v2

nonanon1, 2017-05-11:
I'm going crazy. 
In fresh qmk, I created my own keyboard folder and included a makefile, config and keymap file that I copied from the serial folder and the makefile from the hexwire folder. 

I added rgb controls to the keymap file, and make the necessary changes to the config and makefile. 
I ALSO commented out the master left, and defined ee-hands in the config file. flashing with the two eeprom files always leaves me with everything working if I plug it into the left keyboard, but nothing if I plug it in the right. 

I've tried with previous versions of qmk, and on a hex I was provided by a friend, with which both sides worked as host... but rgb didn't. so at least I know it isn't an issue with my hardware. 

any ideas?

fredizzimo, 2017-05-12:
Did you also flash the EEPROM to both halves as described in the [Setting EE_hands to use either hands as master](https://github.com/qmk/qmk_firmware/blob/3ac9259742bfd428f71c31dbf2bfedace2a7f91b/keyboards/lets_split/readme.md#setting-ee_hands-to-use-either-hands-as-master) section of the readme?

nonanon1, 2017-05-14:
Yes I did. I currently have it working,  but now my rgb isn't working....

closed 2017-06-24 by skullydazed