qmk/qmk_firmware #1295: Macro creator

Dr-Brains, 2017-05-12:
So, this is not really an issue, but I have no idea where to put this...

It's not easy to manually create a macro from a long sentence, that's why I created a Javascript that converts a given string to the QMK macro code.
Right now it supports single line strings with all characters (upper- and lowercase), numbers and symbols (except backslash, dunno how to fix that). It's also not the most elegant code, and could probably be improved by a great margin, but I'm not a coding wizard.

Workflow:
1. Insert typing speed.
2. Insert string to convert.
3. ???
4. PROFIT!
Let me know what you guys think!

If somebody could point me tot the right place to put this, that would be great.

PS: You can test the script by running it in your browsers console or at any Javascript sketch service (eg. labs.codecademy.com)

Code:
```
// INPUT =======================================================================
var inputSpeed = prompt("Please input typing speed in milliseconds", "Speed");
var inputStr = prompt("Please input a string", "String");
var inputArray = inputStr.split("");
var maxLength = inputStr.length;
// CODE ========================================================================
for (var i = 0; i < maxLength; i++)
// lowercase -------------------------------------------------------------------
    if (inputArray[i] === "a")
        inputArray[i] = "T(A)";
    else if(inputArray[i] === "b")
        inputArray[i] = "T(B)";
    else if(inputArray[i] === "c")
        inputArray[i] = "T(C)";
    else if(inputArray[i] === "d")
        inputArray[i] = "T(D)";
    else if(inputArray[i] === "e")
        inputArray[i] = "T(E)";
    else if(inputArray[i] === "f")
        inputArray[i] = "T(F)";
    else if(inputArray[i] === "g")
        inputArray[i] = "T(G)";
    else if(inputArray[i] === "h")
        inputArray[i] = "T(H)";
    else if(inputArray[i] === "i")
        inputArray[i] = "T(I)";
    else if(inputArray[i] === "j")
        inputArray[i] = "T(J)";
    else if(inputArray[i] === "k")
        inputArray[i] = "T(K)";
    else if(inputArray[i] === "l")
        inputArray[i] = "T(L)";
    else if(inputArray[i] === "m")
        inputArray[i] = "T(M)";
    else if(inputArray[i] === "n")
        inputArray[i] = "T(N)";
    else if(inputArray[i] === "o")
        inputArray[i] = "T(O)";
    else if(inputArray[i] === "p")
        inputArray[i] = "T(P)";
    else if(inputArray[i] === "q")
        inputArray[i] = "T(Q)";
    else if(inputArray[i] === "r")
        inputArray[i] = "T(R)";
    else if(inputArray[i] === "s")
        inputArray[i] = "T(S)";
    else if(inputArray[i] === "t")
        inputArray[i] = "T(T)";
    else if(inputArray[i] === "u")
        inputArray[i] = "T(U)";
    else if(inputArray[i] === "v")
        inputArray[i] = "T(V)";
    else if(inputArray[i] === "w")
        inputArray[i] = "T(W)";
    else if(inputArray[i] === "x")
        inputArray[i] = "T(X)";
    else if(inputArray[i] === "y")
        inputArray[i] = "T(Y)";
    else if(inputArray[i] === "z")
        inputArray[i] = "T(Z)";
// uppercase -------------------------------------------------------------------
  else if (inputArray[i] === "A")
      inputArray[i] = "D(LSFT), T(A), U(LSFT)";
  else if(inputArray[i] === "B")
      inputArray[i] = "D(LSFT), T(B), U(LSFT)";
  else if(inputArray[i] === "C")
      inputArray[i] = "D(LSFT), T(C), U(LSFT)";
  else if(inputArray[i] === "D")
      inputArray[i] = "D(LSFT), T(D), U(LSFT)";
  else if(inputArray[i] === "E")
      inputArray[i] = "D(LSFT), T(E), U(LSFT)";
  else if(inputArray[i] === "F")
      inputArray[i] = "D(LSFT), T(F), U(LSFT)";
  else if(inputArray[i] === "G")
      inputArray[i] = "D(LSFT), T(G), U(LSFT)";
  else if(inputArray[i] === "H")
      inputArray[i] = "D(LSFT), T(H), U(LSFT)";
  else if(inputArray[i] === "I")
      inputArray[i] = "D(LSFT), T(I), U(LSFT)";
  else if(inputArray[i] === "J")
      inputArray[i] = "D(LSFT), T(J), U(LSFT)";
  else if(inputArray[i] === "K")
      inputArray[i] = "D(LSFT), T(K), U(LSFT)";
  else if(inputArray[i] === "L")
      inputArray[i] = "D(LSFT), T(L), U(LSFT)";
  else if(inputArray[i] === "M")
      inputArray[i] = "D(LSFT), T(M), U(LSFT)";
  else if(inputArray[i] === "N")
      inputArray[i] = "D(LSFT), T(N), U(LSFT)";
  else if(inputArray[i] === "O")
      inputArray[i] = "D(LSFT), T(O), U(LSFT)";
  else if(inputArray[i] === "P")
      inputArray[i] = "D(LSFT), T(P), U(LSFT)";
  else if(inputArray[i] === "Q")
      inputArray[i] = "D(LSFT), T(Q), U(LSFT)";
  else if(inputArray[i] === "R")
      inputArray[i] = "D(LSFT), T(R), U(LSFT)";
  else if(inputArray[i] === "S")
      inputArray[i] = "D(LSFT), T(S), U(LSFT)";
  else if(inputArray[i] === "T")
      inputArray[i] = "D(LSFT), T(T), U(LSFT)";
  else if(inputArray[i] === "U")
      inputArray[i] = "D(LSFT), T(U), U(LSFT)";
  else if(inputArray[i] === "V")
      inputArray[i] = "D(LSFT), T(V), U(LSFT)";
  else if(inputArray[i] === "W")
      inputArray[i] = "D(LSFT), T(W), U(LSFT)";
  else if(inputArray[i] === "X")
      inputArray[i] = "D(LSFT), T(X), U(LSFT)";
  else if(inputArray[i] === "Y")
      inputArray[i] = "D(LSFT), T(Y), U(LSFT)";
  else if(inputArray[i] === "Z")
      inputArray[i] = "D(LSFT), T(Z), U(LSFT)";
// non-shifted symbols/numbers -------------------------------------------------
else if(inputArray[i] === "1")
    inputArray[i] = "T(1)";
else if(inputArray[i] === "2")
    inputArray[i] = "T(2)";
else if(inputArray[i] === "3")
    inputArray[i] = "T(3)";
else if(inputArray[i] === "4")
    inputArray[i] = "T(4)";
else if(inputArray[i] === "5")
    inputArray[i] = "T(5)";
else if(inputArray[i] === "6")
    inputArray[i] = "T(6)";
else if(inputArray[i] === "7")
    inputArray[i] = "T(7)";
else if(inputArray[i] === "8")
    inputArray[i] = "T(8)";
else if(inputArray[i] === "9")
    inputArray[i] = "T(9)";
else if(inputArray[i] === "0")
    inputArray[i] = "T(0)";
else if(inputArray[i] === "-")
    inputArray[i] = "T(MINS)";
else if(inputArray[i] === "=")
    inputArray[i] = "T(EQL)";
else if (inputArray[i] === ",")
    inputArray[i] = "T(COMM)";
else if(inputArray[i] === ".")
    inputArray[i] = "T(DOT)";
else if(inputArray[i] === "/")
    inputArray[i] = "T(SLSH)";
else if(inputArray[i] === ";")
    inputArray[i] = "T(SCLN)";
else if(inputArray[i] === "'")
    inputArray[i] = "T(QUOT)";
else if(inputArray[i] === "[")
    inputArray[i] = "T(LBRC)";
else if(inputArray[i] === "]")
    inputArray[i] = "T(RBRC)";
/*
BACKSLASH: this line gives an error, dunno how to fix it...
else if(inputArray[i] === "\")
    inputArray[i] = "T(BSLS)";
*/
else if(inputArray[i] === " ")
    inputArray[i] = "T(SPC)";
// shifted symbols -------------------------------------------------------------
else if(inputArray[i] === "!")
    inputArray[i] = "D(LSFT), T(1), U(LSFT)";
else if(inputArray[i] === "@")
    inputArray[i] = "D(LSFT), T(2), U(LSFT)";
else if(inputArray[i] === "#")
    inputArray[i] = "D(LSFT), T(3), U(LSFT)";
else if(inputArray[i] === "$")
    inputArray[i] = "D(LSFT), T(4), U(LSFT)";
else if(inputArray[i] === "%")
    inputArray[i] = "D(LSFT), T(5), U(LSFT)";
else if(inputArray[i] === "^")
    inputArray[i] = "D(LSFT), T(6), U(LSFT)";
else if(inputArray[i] === "&")
    inputArray[i] = "D(LSFT), T(7), U(LSFT)";
else if(inputArray[i] === "*")
    inputArray[i] = "D(LSFT), T(8), U(LSFT)";
else if(inputArray[i] === "(")
    inputArray[i] = "D(LSFT), T(9), U(LSFT)";
else if(inputArray[i] === ")")
    inputArray[i] = "D(LSFT), T(0), U(LSFT)";
else if(inputArray[i] === "_")
    inputArray[i] = "D(LSFT), T(MINS), U(LSFT)";
else if(inputArray[i] === "+")
    inputArray[i] = "D(LSFT), T(EQL), U(LSFT)";
else if (inputArray[i] === "<")
    inputArray[i] = "D(LSFT), T(COMM), U(LSFT)";
else if(inputArray[i] === ">")
    inputArray[i] = "D(LSFT), T(DOT), U(LSFT)";
else if(inputArray[i] === "?")
    inputArray[i] = "D(LSFT), T(SLSH), U(LSFT)";
else if(inputArray[i] === ":")
    inputArray[i] = "D(LSFT), T(SCLN), U(LSFT)";
else if(inputArray[i] === '"')
    inputArray[i] = "D(LSFT), T(QUOT), U(LSFT)";
else if(inputArray[i] === "{")
    inputArray[i] = "D(LSFT), T(LBRC), U(LSFT)";
else if(inputArray[i] === "}")
    inputArray[i] = "D(LSFT), T(RBRC), U(LSFT)";
else if(inputArray[i] === "|")
    inputArray[i] = "D(LSFT), T(BSLS), U(LSFT)";
// OUTPUT ======================================================================
var output = inputArray.toString();
var frontMacro = "return MACRO( I(" + inputSpeed + "), ";
var backMacro = ", END  );";
var finalMacro = frontMacro.concat(output, backMacro);
console.log(finalMacro);
```

Traverse, 2017-05-20:
@Dr-Brains I had some spare time and decided to improve on your idea. 

Hope you like it: https://codepen.io/Traverse/full/zwMwKX/

Dr-Brains, 2017-05-21:
@Traverse that is exactly like I envisioned it! Looks great and works great, good job! Thanks for improving my meager attempt at JS :)

Traverse, 2017-05-21:
@Dr-Brains Thanks, your idea was good I just improved it a bit and made a GUI. Anyway small projects like these are great for learning so just keep doing stuff like this and you'll become a coding wizard in no time as well. 😉

Dr-Brains, 2017-06-09:
@Traverse I posted this on the olkb Reddit: [go say hi!](www.reddit.com/r/olkb/comments/6g9gga/qmk_macrofier_tool/)

skullydazed, 2017-06-24:
This was a neat idea, but I'm not sure what we can do with this issue anymore. I'd encourage the two of you to post what you have to its own GitHub repo so others can find and benefit from your work.

closed 2017-06-24 by skullydazed