qmk/qmk_firmware #1296: kc60v2 Works on Linux but shows up as ATm32U4DFU on Windows 10

r0n22, 2017-05-12:
Currently working on an issue with my keyboard (kc60v2) using the default wigguno layout.

It will connect to my Windows 10  computer but after about 2-3 hours (Usually happens during multi key press) keyboard looses connection and re installs on windows as ATm32U4DFU.

Has anyone experienced something similar?  I am using the latest qmk version.

shieldsd, 2017-05-12:
Are you accidentally hitting a key combination that puts it in dfu mode?

fredizzimo, 2017-05-12:
I was going to say the same thing, it will show up as that when the keyboard is in flashing mode.

And looking at that keymap, it seems to be very easy to be hit. The reset button is the same button as left ctrl, but on the function layer.

sgoodwin, 2017-05-12:
Yeah, I maybe didn't pick a great place for the reset button. My bad (I made the default layout). Feel free to make a new layout without this issue.

r0n22, 2017-05-13:
So I have removed the reset from the keyboard layout and done a couple restarts of my computer but Windows 10 still shows the keyboard as ATm32U4DFU

fredizzimo, 2017-05-13:
Another thing that could cause the keyboard to enter the bootloader mode is if the firmware crashes. 

But it's hard to see how windows could do that, while Linux wouldn't. A couple of things comes to my mind, the CAPS, NUM and SCROLL lock leds are sent by windows, so try to see if you can cause crashes by enabling or disabling those.

The other thing is the sysex protocol, so if you have `API_SYSEX_ENABLE`, try to disable that.

fredizzimo, 2017-05-13:
Ah I remember now that just the midi support could potentially cause problems, see https://github.com/qmk/qmk_firmware/issues/1011.

So try setting `MIDI_ENABLE=no` too.

r0n22, 2017-05-13:
So I did not realize that the DFU mode was persistence across unplugs and re-plugs.  I have removed the reset key and will close as that seems to be the most logical cause of the issue.

closed 2017-05-13 by r0n22