qmk/qmk_firmware #1304: Really odd behavior with LT()

gordonlew, 2017-05-17:
I am trying to implement LT(4, KC_RALT). My fourth layer is a number layer. When I put RALT in and try to raise the layer, instead of numbers I get numbers plus shift, i.e. &*() instead of 7980. When I instead put LALT I get shift plus alt, i.e ‡°·‚ instead of 7890. What's up with that and how can I get this to work?

fredizzimo, 2017-05-24:
What are you trying to do? Combing LT with modifiers doesn't really make sense, since when held it would switch layers, and when tapped send the modifier, which would immediately get released. 

I guess you want it the other way? As far as I know there's no ready function for it, you would have to make a macro yourself. See the [macro documentation](https://docs.qmk.fm/Macros.html), especially the "Example: Single-key copy/paste (hold to copy, tap to paste)", which shows how to make a custom dual role key.

gordonlew, 2017-05-24:
Yes, in general it doesn't make sense to tap modifiers, but in Excel, alt functions as a leader key, so it makes sense then to tap it. Thanks for the tip, though! I'll look into whether a custom macro would give me this functionality.

skullydazed, 2017-06-24:
I'm closing this to clean up our open issues, but if you still have questions @gordonlew please reopen this or open a new issue.

closed 2017-06-24 by skullydazed

jtojnar, 2022-03-26:
Sorry for bumping, I found this thread when searching for the other way round. Apparently, that feature is now implemented and called [Mod Tap](https://docs.qmk.fm/#/mod_tap).