qmk/qmk_firmware #1310: Getting completly wrong keypresses

samuelcolles, 2017-05-19:
I am trying to make my own keyboard project based off of the fivethirteen handwired layout, but after i compiled and loaded the project (using vagrant and teensy loader) while testing the keyboard i am getting completely wrong key presses. So i then tried compiling and using the onkey project, and i was having the same problem. when i actuate the button instead of getting the 'a' key (clearly indicated in keymap.c) i am getting the 'apps' key instead.  Any help or suggestions would be greatly appreciated.

fredizzimo, 2017-05-24:
It sounds like you have either wired the it wrong, defined the matrix `KEYMAP` macro wrong, or perhaps the `MATRIX_COL_PINS` or `MATRIX_ROW_PINS`. Maybe the columns and rows are swapped?

samuelcolles, 2017-05-25:
I dont think so. In the onekey there is only 1 key so only one row and one col. Thanks for the response though, im going to order another teensy to see if that is the problem(I did reuse it from another project).

fredizzimo, 2017-05-25:
Yes, I missed the fact that you tested with the onekey version. What key do you have specified in the keymap and what do you get for that? Is there some pattern, if you try diffrent keys?

I really doubt your teensy could be the problem, as it's very unlikely it would be able to send any keypresses at  all if it's broken. 

But the only other thing I can think of right now is that you have some really strange keymap on the operating system side.

drashna, 2018-03-25:
This issue was either solved or abandoned.

Closing issue.  If needed, please reopen or open a new issue.

closed 2018-03-25 by drashna