qmk/qmk_firmware #1312: Failing to compile infinity ergodox on ArchLinux

paxy97, 2017-05-20:
I've been trying for a while now to compile for the ergodox after pulling the newest changes without any luck. This is the error I get, even on a completely clean clone.

```
Compiling: keyboards/ergodox/infinity/drivers/gdisp/st7565ergodox/gdisp_lld_ST7565.c               In file included from ./lib/ugfx/src/gdisp/gdisp_driver.h:22:0,
                 from keyboards/ergodox/infinity/drivers/gdisp/st7565ergodox/gdisp_lld_ST7565.c:14:
./lib/ugfx/src/gdisp/../gdriver/gdriver.h:90:33: error: duplicate 'const' declaration specifier [-Werror=duplicate-decl-specifier]
 typedef const struct GDriverVMT const GDriverVMTList[1];
                                 ^~~~~
In file included from keyboards/ergodox/infinity/drivers/gdisp/st7565ergodox/gdisp_lld_ST7565.c:14:0:
./lib/ugfx/src/gdisp/gdisp_driver.h:731:17: error: duplicate 'const' declaration specifier [-Werror=duplicate-decl-specifier]
  const GDISPVMT const GDISP_DRIVER_VMT[1] = {{
                 ^~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [tmk_core/rules.mk:361: .build/obj_ergodox_infinity/drivers/gdisp/st7565ergodox/gdisp_lld_ST7565.o] Error 1
Make finished with errors
make: *** [../../Makefile:508: infinity-default-dfu-util] Error 1
```

I started throwing this error since yesterday when I pulled the lates changes. The last pull was maybe a week or two before that. Checking blame on the two mentioned files shows that the last changes were in 2014, so something else seems to be the problem.

I'm running ArchLinux with the latest updates. Here are the versions of the toolchain used:

```
arm-none-eabi-binutils 2.28-1
arm-none-eabi-gcc 7.1.0-1
arm-none-eabi-gdb 7.12.1-2
arm-none-eabi-newlib 2.5.0.20170421-1
```

paxy97, 2017-05-23:
The problem seems to be because of the newer version of gcc. After downgrading to 6.3 it works just fine.

fredizzimo, 2017-05-24:
I'm reopening this, more people will soon upgrade to Gcc 7, so this has to be fixed.

I noticed this myself a couple of weeks ago when I was compiling with clang. The bug is not really in QMK itself, but in the external uGfx library. It's possible that it's fixed in the latest version and the issue is a matter of upgrading. Otherwise I will send a pull request there first.

fredizzimo, 2017-06-04:
I have now opened the pull request #1363, which should fix these issues.

PR #1363 (2017-06-14): Update uGFX and Chibios submodules + fixes
The submodules now point to the QMK Github.

Also update uGfx to version 2.7, with some custom fixes.

Fixes #1312, by removing duplicate const declarations
The other fixes are for the upcoming emulator support.
closed 2017-06-14 by skullydazed