qmk/qmk_firmware #1313: Document config.h options and effects

haasn, 2017-05-21:
As far as I can tell, there's no place to get information about available `config.h` options and their effects other than reading the source code.

For example, I recently figured out that I can seemingly use config.h changes to solve **two** of the annoying bugs I had with qmk_firmware (`#define PREVENT_STUCK_MODIFIERS` and `#undef IGNORE_MOD_TAP_INTERRUPT`, respectively); and I only found out about these while reading other people's various `config.h` files.

It's annoying that I can't really find out more information about what they actually do, other than a vague guess based on the title. If there is documented at all in the repo or the website, I cannot find it.

I would probably suggest adding a commented-out version of all available options + documentation to the “config.h” template.

jackhumbert, 2017-05-21:
PREVENT_STUCK_MODIFIERS is described [here](https://github.com/qmk/qmk_firmware_docs/blob/master/Keymap-examples.md) - this isn't on the docs site yet because that page isn't linked in the summary. We're still doing a lot of organising there :)

jayliu50, 2017-06-29:
A small plug for a request to document the ability to redefine magic modifier for magic key command, as accidentally activating it has tripped up some of us, myself included  :)  (#1233) 

```c
/* key combination for magic key command */
#define IS_COMMAND() ( \
    keyboard_report->mods == (MOD_BIT(KC_LSHIFT) | MOD_BIT(KC_RSHIFT)) \
)
```

commit 67cc5ceb (2017-10-25): Restructure the hardware and feature docs to make things easier to find (#1888)

* fix #1313 by documenting more config.h options

* Clean up and organize documentation
closed 2017-10-25 by skullydazed