qmk/qmk_firmware #1314: Build fail on Arch latest avr-gcc

sQVe, 2017-05-22:
Using the latest `avr-gcc (7.1.0-1)` I'm getting the following error when trying to compile:

```
% make sqve                                                                                                                                                                                                                                                                       ~/code/qmk_firmware/keyboards/ergodox master
Making ergodox/ez with keymap sqve

avr-gcc (GCC) 7.1.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  22752	      0	  22752	   58e0	ergodox_ez_sqve.hex

Compiling: keyboards/ergodox/keymaps/sqve/keymap.c                                                  [OK]
Compiling: keyboards/ergodox/ez/ez.c                                                               keyboards/ergodox/ez/ez.c: In function 'init_mcp23018':
keyboards/ergodox/ez/ez.c:60:24: error: increment of a boolean expression [-Werror=bool-operation]
         i2c_initialized++;
                        ^~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [tmk_core/rules.mk:361: .build/obj_ergodox_ez_sqve/keyboards/ergodox/ez/ez.o] Error 1
Make finished with errors
make: *** [../../Makefile:508: sqve] Error 1
```

Downgrading to version `6.3.0-1` solves the issue.

fredizzimo, 2017-05-24:
Gcc 7 enabled that warning. I guess because C++17 forbids incrementing boolean values, so it makes sense to give a warning in C as well. Since many users soon will upgrade, I think this should be fixed ASAP.

The fix is easy, just change `i2c_initialized++` to `i2c_initialized=true`. Or maybe better, remove the variable and move the code to `matrix_init_keyboard`. Even if the changes are trivial I can't send a pull request myself, since I don't own an EZ.

ariasuni, 2017-05-24:
After fixing that first error, the compiler couldn’t find the function that set on or off the led. I didn’t find how to compile it so I reverted to avr-gcc 6.3.

fredizzimo, 2017-05-25:
@ariasuni, do you remember the exact error that you got? I can probably help with getting it to compile.

ariasuni, 2017-05-26:
```
Linking: .build/ergodox_ez_ariasuni.elf                                                             [ERRORS]                                    
 |                                                                                                                                              
 | .build/obj_ergodox_ez_ariasuni/keyboards/ergodox/ez/ez.o : Dans la fonction « ergodox_blink_all_leds » :                                     
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/ez.c:32 : référence indéfinie vers « ergodox_led_all_off »                          
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/ez.c:33 : référence indéfinie vers « ergodox_led_all_set »                          
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/ez.c:34 : référence indéfinie vers « ergodox_right_led_1_on »                       
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/ez.c:36 : référence indéfinie vers « ergodox_right_led_2_on »
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/ez.c:38 : référence indéfinie vers « ergodox_right_led_3_on »
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/ez.c:40 : référence indéfinie vers « ergodox_right_led_1_off »
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/ez.c:42 : référence indéfinie vers « ergodox_right_led_2_off »
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/ez.c:44 : référence indéfinie vers « ergodox_right_led_3_off »
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/ez.c:47 : référence indéfinie vers « ergodox_led_all_off »
 | .build/obj_ergodox_ez_ariasuni/keyboards/ergodox/ez/keymaps/ariasuni/keymap.o : Dans la fonction « matrix_init_user » :
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/keymaps/ariasuni/keymap.c:162 : référence indéfinie vers « ergodox_led_all_set »
 | .build/obj_ergodox_ez_ariasuni/keyboards/ergodox/ez/keymaps/ariasuni/keymap.o : Dans la fonction « matrix_scan_user » :
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/keymaps/ariasuni/keymap.c:170 : référence indéfinie vers « ergodox_right_led_1_on »
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/keymaps/ariasuni/keymap.c:173 : référence indéfinie vers « ergodox_right_led_2_on »
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/keymaps/ariasuni/keymap.c:176 : référence indéfinie vers « ergodox_right_led_3_on »
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/keymaps/ariasuni/keymap.c:171 : référence indéfinie vers « ergodox_right_led_1_off »
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/keymaps/ariasuni/keymap.c:174 : référence indéfinie vers « ergodox_right_led_2_off »
 | /home/ariasuni/projets/qmk_firmware/keyboards/ergodox/ez/keymaps/ariasuni/keymap.c:177 : référence indéfinie vers « ergodox_right_led_3_off »
 | collect2: error: ld returned 1 exit status
```

fredizzimo, 2017-05-26:
Since I'm a C++ guy, I had to do a bit reading about how things works in C. My conclusion from that, is that `static inline` in header files is always wrong. Instead you should have the function definition in the header file, and an external declartation in one source file. `Static inline` should be used for inline functions that are used from just one translation unit, and in that case, defined in the source  file.

So in the `Ez.h`file:
```C
inline void ergodox_board_led_on(void) { DDRD |= (1<<6); PORTD |= (1<<6); }
```
And in the `Ez.c`file:
```C
extern inline void ergodox_board_led_on(void);
```

So if you change the LED functions like that, I think it's all good for a pull request.

ariasuni, 2017-05-26:
Sorry, the declaration should rather be in `ez.h` and definition in `ez.c`, isn’t it?

fredizzimo, 2017-05-26:
No the definition (implementation) in the header file, so that all translation files can see and inline it. The declaration with extern is just to give a hint to the linker, so that it knows in which object file it should put the symbol.

fredizzimo, 2017-05-27:
Just out of curiosity, did you check how the new changes compares in terms of hex sizes?

ariasuni, 2017-05-29:
@fredizzimo well all the sizes I measured are here: https://github.com/qmk/qmk_firmware/issues/1314#issuecomment-304165495

I didn’t compare for other keymap but you can try with my fork.

coderkun, 2017-06-18:
Any news on this?

sQVe, 2017-06-19:
@coderkun This should be fixed when PR #1331 is merged, as far as I know. Until then just run version `6.3.0`.

PR #1331 (2017-06-22): Fix undefined reference errors with avr-gcc 7.1
Fix #1314
closed 2017-06-22 by ezuk