qmk/qmk_firmware #1315: Custom keycode becomes KC_NO

Maartenwut, 2017-05-22:
I'm trying to make some custom keycodes, but it seems that the first one overwrites KC_NO for some reason. Here's my keymap with the custom keycodes:

```
#include "tada68.h"

// Each layer gets a name for readability, which is then used in the keymap matrix below.
// The underscores don't mean anything - you can have a layer called STUFF or any other name.
// Layer names don't all need to be of the same length, obviously, and you can also skip them
// entirely and just use numbers.

#define _MA 0
#define _FL 1

#define TRNS KC_TRNS

enum emoticons {
	LENNY,
	DWNHRT,
	SHRUG
};

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
  /* Keymap _MA: Main Layer
   * ,----------------------------------------------------------------.
   * |Esc | 1|  2|  3|  4|  5|  6|  7|  8|  9|  0|  -|  =|Backsp | MP |
   * |----------------------------------------------------------------|
   * |Tab  |  Q|  W|  E|  R|  T|  Y|  U|  I|  O|  P|  [|  ]|  \  |Del |
   * |----------------------------------------------------------------|
   * |CAPS   |  A|  S|  D|  F|  G|  H|  J|  K|  L|  ;|  '|Return |PgUp|
   * |----------------------------------------------------------------|
   * |Shift   |  Z|  X|  C|  V|  B|  N|  M|  ,|  .|  /|Shift | Up|PgDn|
   * |----------------------------------------------------------------|
   * |Ctrl|Win |Alt |        Space          |Alt|Ctrl| FN|Lef|Dow|Rig |
   * `----------------------------------------------------------------'
   */
[_MA] = KEYMAP_ANSI(
  KC_ESC, KC_1, KC_2, KC_3, KC_4, KC_5, KC_6, KC_7, KC_8, KC_9, KC_0, KC_MINS, KC_EQL,     KC_BSPC,    KC_MPLY, \
  KC_TAB, KC_Q, KC_W, KC_E, KC_R, KC_T, KC_Y, KC_U, KC_I,  KC_O,  KC_P, KC_LBRC,KC_RBRC,KC_BSLS,       KC_DEL, \
  KC_CAPS, KC_A, KC_S,   KC_D, KC_F, KC_G, KC_H, KC_J, KC_K,  KC_L,  KC_SCLN, KC_QUOT,  KC_ENT,        KC_PGUP, \
  KC_LSFT, KC_Z, KC_X,   KC_C, KC_V, KC_B, KC_N, KC_M, KC_COMM, KC_DOT,  KC_SLSH, KC_RSFT,    KC_UP,   KC_PGDN, \
  KC_LCTL, KC_LGUI, KC_LALT,              KC_SPC,      KC_RALT, KC_RCTRL, MO(_FL),  KC_LEFT, KC_DOWN, KC_RGHT),

  /* Keymap _FL: Function Layer
   * ,----------------------------------------------------------------.
   * |   | F1|F2 |F3 |F4 |F5 |F6 |F7 |F8 |F9 |F10|F11|F12|       |Psc |
   * |----------------------------------------------------------------|
   * |Lenny|MbL|MsU|MbR|   |   |   |   |   |   |   |   |   |     |    |
   * |----------------------------------------------------------------|
   * |DwnHrt|MsL|MsD|MsR|   |   |   |   |   |   |   |   |        |Hme |
   * |----------------------------------------------------------------|
   * |  Shrug |   |   |   |   |   |   |   |   |   |   |      |MwU|End |
   * |----------------------------------------------------------------|
   * |    |    |    |                       |   |   |    |MwL|MwD|MwR |
   * `----------------------------------------------------------------'
   */
[_FL] = KEYMAP_ANSI(
  KC_GRV, KC_F1,  KC_F2,  KC_F3,  KC_F4,  KC_F5,  KC_F6,  KC_F7,  KC_F8,  KC_F9,  KC_F10,  KC_F11,  KC_F12,             TRNS, KC_PSCR, \
  LENNY, KC_BTN1, KC_MS_U,  KC_BTN2, TRNS, TRNS, TRNS, TRNS, TRNS, TRNS, TRNS, TRNS, TRNS,TRNS, TRNS,   \
  DWNHRT, KC_MS_L, KC_MS_D, KC_MS_R,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,                TRNS,KC_HOME, \
  SHRUG, TRNS, TRNS, TRNS, TRNS,TRNS, TRNS,TRNS,TRNS,TRNS,TRNS,TRNS,             KC_WH_U, KC_END, \
  TRNS, TRNS, TRNS,                       KC_NO,                   TRNS,TRNS,TRNS,   KC_WH_L,KC_WH_D, KC_WH_R),
};

void tap(uint16_t keycode){
    register_code(keycode);
    unregister_code(keycode);
};

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
	switch (keycode) {
		case LENNY:
			if(record->event.pressed){
				set_unicode_input_mode(UC_WIN);
				register_code(KC_LSFT);
				tap(KC_9);                                      // Head
				unregister_code(KC_LSFT);
				tap(KC_SPC);                                    // Space
				process_unicode((0x0361|QK_UNICODE), record);   // Eyebrow
				process_unicode((0x00B0|QK_UNICODE), record);   // Eye
				tap(KC_SPC);
				process_unicode((0x035C|QK_UNICODE), record);   // Mouth
				process_unicode((0x0296|QK_UNICODE), record);   // Nose
				tap(KC_SPC);
				process_unicode((0x0361|QK_UNICODE), record);   // Eyebrow
				process_unicode((0x00B0|QK_UNICODE), record);   // Eye
				register_code(KC_LSFT);
				tap(KC_0);                                      // Head
				unregister_code(KC_LSFT);
			}
			return false;
			break;
		case DWNHRT:
			if(record->event.pressed){
				set_unicode_input_mode(UC_WIN);
				register_code(KC_LSFT);
				tap(KC_9);                                      // Head
				unregister_code(KC_LSFT);
				process_unicode((0x00B4|QK_UNICODE), record);   // Eyebrow
				process_unicode((0x30FB|QK_UNICODE), record);   // Eye
				process_unicode((0x03C9|QK_UNICODE), record);   // Mouth
				process_unicode((0x30FB|QK_UNICODE), record);   // Eye
				process_unicode((0x0060|QK_UNICODE), record);   // Eyebrow
				register_code(KC_LSFT);
				tap(KC_0);                                      // Head
				unregister_code(KC_LSFT);
			}
			return false;
			break;
		case SHRUG:
			if(record->event.pressed){
				set_unicode_input_mode(UC_WIN);
				process_unicode((0x00AF|QK_UNICODE), record);   // Hand
				tap(KC_BSLS);                                   // Arm
				register_code(KC_LSFT);
				tap(KC_UNDS);                                   // Arm
				tap(KC_LPRN);                                   // Head
				unregister_code(KC_LSFT);
				process_unicode((0x30C4|QK_UNICODE), record);   // Face
				register_code(KC_LSFT);
				tap(KC_RPRN);                                   // Head
				tap(KC_UNDS);                                   // Arm
				unregister_code(KC_LSFT);
				tap(KC_SLSH);                                   // Arm
				process_unicode((0x00AF|QK_UNICODE), record);   // Hand
			}
			return false;
			break;
	}
	return true;
};
```

As one can see, my space bar, mapped to KC_NO on the function layer, outputs the keycode LENNY. This is not the behavior I expected. When adding even more keycodes, keys like A get replaced with my own.

jackhumbert, 2017-05-22:
Your `enum` needs to start with `SAFE_RANGE`, like [this](https://github.com/qmk/qmk_firmware/blob/master/keyboards/planck/keymaps/default/keymap.c#L29) :)

Maartenwut, 2017-05-22:
Thank you Jack, this solved the problem! Maybe write a little about it in the wiki for new users?

closed 2017-05-22 by Maartenwut

jackhumbert, 2017-05-22:
Yeap! All that's coming soon :) btw, new docs are at https://docs.qmk.fm!

fnurl, 2017-10-02:
Man, this had me stumped for quite some time. I thought the SAFE_RANGE had to do with having multiple default layouts (QWERTY, DVORAK, COLEMAK), so I removed it  and it worked just fine until recently when I added more custom keycodes..

kulukimak, 2018-10-21:
Could someone clarify what exactly SAFE_RANGE does and why it has to be added? I was having the same error and the advice fixed my problem, but I don't know why.

noroadsleft, 2018-10-21:
@kulukimak,

Think of the keycode space like a free-standing set of dresser drawers, where each drawer can hold one keycode.

QMK adds keycodes to the firmware based on which features are enabled in the keyboard/keymap. However, there's only so many drawers in the dresser. Because different boards and keymaps may have different features, one keymap might have (throwing random numbers around) 121 keycodes, and another might have 140. `SAFE_RANGE` is the first available space that doesn't have a keycode assigned, which is why when you create macro keys, you need to use `SAFE_RANGE`. Otherwise, your macro keycodes may overwrite keycodes that already exist, depending on the "drawers" that are available.