qmk/qmk_firmware #1320: Variable tapping term for tap dance.

nikchi, 2017-05-25:
I've implemented a solution to have variable tapping terms when using ` ACTION_TAP_DANCE_FN_ADVANCED` by using the null value that is unused in the advanced tap dance functions, but it involves changing how users call it. 

Also unsure how the changes made in `matrix_scan_tap_dance()` will affect the tap dance doubles, as I cast user_data to an int, but the double function has something else in it.

Looking for input on:    
1. is this something that might get merged,  
2. does it interfere with current implementation of `ACTION_TAP_DANCE_DOUBLE`,  
3. is there some better way to implement this, where user_data will be multiple kinds of values (ie struct)

e695b5a33b97cfb4f9dd8bc8ecaff8aa7e0f14cc

fredizzimo, 2017-05-25:
I think the idea is good, but I don’t like the usage of `user_data`, since it's meant for other things, and as you said, I'm pretty sure it's broken for the doubles.

Instead I think we have to accept 2 bytes exta for the custom term, by adding a `uint16_t` field for it. Technically 8 bit could be enough if multiplying with 10 for example. But due to struct padding that would not save any memory, unless using custom attributes, which in turn could increase the code size.

Also instead of changing the existing macros you could make a new set, like `ACTION_TAP_DANCE_FN_ADVANCED_T` where `T` stands for term or time. They would otherwise be the same, but take time as the last parameter.

nikchi, 2017-05-25:
Cool, PR created.

I've moved the functionality to a new macro, and also added a uint16_t  to the action struct.

closed 2017-05-30 by nikchi