qmk/qmk_firmware #1324: Question: Initialization without USB handshake

yiancar, 2017-05-26:
It seem that if you just power the keyboard without a usb connection, matrix_scan() is not entered, ie the keyboard sleeps?

Is there a way to disable this? 
I already tried NO_SUSPEND_POWER_DOWN.
This is for a split keyboard project and therefor the slave keyboard needs to be connected with a usb to send the matrix over. If I unplug the usb from the slave it keeps functioning so I presume there is a usb check at startup?

jackhumbert, 2017-05-26:
Yeah, I think that's [this loop](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/lufa/lufa.c#L1184) - I had disabled it as a test when using the keyboard with a wall wart, and it worked, but that was last year or so. I've been meaning to add a flag (similar to the bluetooth one) that allows you to disable it :)

yiancar, 2017-05-26:
Hello,
Thank you so much, yes that fixed it. A flag would be appreciated

closed 2017-05-27 by yiancar