qmk/qmk_firmware #1330: Question: trying to override the quantum matrix_scan and matrix_init

realSpok, 2017-05-26:
I am currently building a custom split keyboard, based on a let's split design but wireless (bluefruit ez-key with rf modules between the halves for maximum wirelessness). 
However when I try to use a custom matrix_init and matrix_scan defined in a new matrix.c (inspired by the ergodox way), the make fails at linking time, complaining that those functions already exist in quantum. Shouldn't those be declared weak ? However, it doesn't seem to cause a problem for building the ergodox ? Is there something I am missing ? 
Thanks for any tip.

realSpok, 2017-05-26:
I should add that I am using a teensy 2.0 in the master half.

fredizzimo, 2017-05-26:
You need to add `CUSTOM_MATRIX=yes` to your `rules.mk` file.

realSpok, 2017-05-27:
This worked ! Thanks a lot for the help. I am very grateful.

closed 2017-05-27 by realSpok