qmk/qmk_firmware #1332: Nkey rollover

packman86, 2017-05-27:
I am noticing that Nkey rollover is not enabled on my Ergodox Infinity.
How do enable it again?

fredizzimo, 2017-05-27:
It should be enabled by default and the Makefile confiuration variable is `NKRO_ENABLE`, so if you have a custom Makefile for your keymap, check that it's not set to false.

Otherwise check the [FAQ](https://docs.qmk.fm/FAQ.html), and the `NKRO Doesn't work` section.

g3rain1, 2017-05-27:
How can we set it on by default instead of needing to toggle it on after every power on? It's already set to enabled in the make file.

fredizzimo, 2017-05-27:
It should be enabled when the operating system requests the right HID protocol. See [ChibiOs](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/chibios/usb_main.c#L949). However I didn't see something similar for LUFA, so it might be buggy. I will investigate it further to see how its actually supposed to work.

It used to be controlled by the EECONFIG, but that doesn't seem to be the case anymore.

Anyway, there's a way to force it, `FORCE_NKRO=yes` in the makefile. But be aware that the keyboard might not work in the Bios with that.

fredizzimo, 2017-05-27:
Also see #555 (Why is NKRO disabled by default)

packman86, 2017-05-29:
I was able to get it working by adding the NKRO_ENABLE and also hitting lshift-rshift-n.

closed 2017-05-29 by packman86