qmk/qmk_firmware #1333: Switching layer makes TT think it's been tapped.

Maartenwut, 2017-05-28:
On my keyboard, I have an FN key mapped to TT(_FN). When I tap it 5 times, it switches to the FN layer and when I hold it, it momentarily switches to the FN layer. On this FN layer, I have a key mapped to TO(_LO), the 'locked' layer. This 'locked' layer has one key to momentarily switch to the 'unlock' layer (MO(_UL)), which has a key to go back the the main layer (TO(_MA)). But when I go back to the main layer, it actually goes to the FN layer and I'd have to press the FN key 5 times to go back to the main layer. This behavior is not present when I use MO instead of TT.

Here's my keymap: https://plakbak.nl/5AYk484ZOp

fredizzimo, 2017-06-10:
I haven't verified it by looking at the code, but if the documentation is right, them the issue is that `TT` is a layer toggle, and not a switch, so acutally both the main and the function layers become active.

Maartenwut, 2017-06-11:
But it shouldn't toggle when coming from another layer, right?

drashna, 2018-03-25:
I could be wrong, but I think that the TO() keycode will clear the TT() layer fro the stack.  

Also, did you ever get this resolved?

closed 2019-12-28 by drashna