qmk/qmk_firmware #1345: 1 push = all keycode of the col

Dbroqua, 2017-05-29:
Bad title...

I'm trying to realize my own keyboard and I'm a little disappointed...

After soldering all the parts and flashing the keyboard with QMK, all keycodes work except those of a row (f/j row).

Example:
If I press key `4`, computer receive `4`.
If I press key `f`, computer receive `4rfv`...

All diode are connected on the rows and in my `config.h` I wrote `#define DIODE_DIRECTION COL2ROW`.

I'm not sure if the problem is on my PCB, on my soldering or in the wrong configuration of my QMK directory.

Do you have any idea? 

It's really strange :(

Dbroqua, 2017-05-29:
My configuration can be found here: https://github.com/Dbroqua/qmk_firmware/tree/feature/DK60/keyboards/dk60

Regards,
Damien

jackhumbert, 2017-05-29:
That usually happens when there's a short somewhere on the `f` key - does it happen anywhere else on the keyboard? If not, it sounds like it'd be hardware-related.

Dbroqua, 2017-05-29:
All the keys of the row have the same issue (a, s, d, f, g,.....).

jackhumbert, 2017-05-29:
It may be that row that's shorted to GND then - if you can attach photos, we might be able to see where the short is.

Dbroqua, 2017-05-29:
Ok,
Now I know what I need to search :D

Dbroqua, 2017-05-29:
And... It's that :'( .

Thank you!

Dbroqua, 2017-05-29:
RED: no separation between GND/Row2
BLUE: problem solved! :|

[image]

https://cloud.darkou.fr/index.php/s/ezEumjEV5QD5qCJ

Thanks!

closed 2017-05-29 by Dbroqua