qmk/qmk_firmware #1346: CAPS_LOCK led

Dbroqua, 2017-05-30:
Hello,

I'm trying to flash my own keyboard and I have a new problem.

On my keyboard I've define 2 leds (CAPS LOCK in E6 and ESC_LED on F0). 

I've copied the GH60 config to implement leds in my keyboard.

But... it's not working :( .

My code can be found here: https://github.com/Dbroqua/qmk_firmware/tree/feature/DK60/keyboards/dk60

If I use other method like this : 
dk60.c:
```/
void matrix_init_kb(void) {
	led_init_ports();
};

void led_init_ports(void) {
    // * Set our LED pins as output
    DDRE |= (1<<6);
}

void led_set_kb(uint8_t usb_led) {
    if (usb_led & (1<<USB_LED_CAPS_LOCK)) {
        // Turn capslock on
        PORTE &= ~(1<<6);
    } else {
        // Turn capslock off
        PORTE |= (1<<6);
    }
}
```
Led is on when caps lock is disabled and off when caps lock is enabled.

It's strange but the led is ok!

Do you have any idea ?

I think I've miss one config but which one :/

Regards,
Damien

Maartenwut, 2017-05-30:
Why don't you just switch it around then?

```
void matrix_init_kb(void) {
	led_init_ports();
};

void led_init_ports(void) {
    // * Set our LED pins as output
    DDRE |= (1<<6);
}

void led_set_kb(uint8_t usb_led) {
    if (usb_led & (1<<USB_LED_CAPS_LOCK)) {
        // Turn capslock on
        PORTE |= (1<<6);
    } else {
        // Turn capslock off
        PORTE &= ~(1<<6);
    }
}
```

Dbroqua, 2017-05-30:
Hello,

Because I want to be able to activate my 2 leds via actions like activate Esc led when I push en Fn button or other.

Your solution is ok for CAPS_LED but for ESC_LED...

Regards,
Damien

fredizzimo, 2017-05-30:
Add this to configure the output

```c
void led_init_ports(void) {
    // * Set our LED pins as output
    DDRE |= (1<<6);
    DDRF |= (1<<0);
}
```

And then this for the actual output
```
inline void dk60_caps_led_on(void)      {  PORTE |= (1<<6); }
inline void dk60_esc_led_on(void)    	{  PORTF |= (1<<0); }

inline void dk60_caps_led_off(void)     { PORTE &= ~(1<<6); }
inline void dk60_esc_led_off(void)   	{ PORTF &= ~(1<<0); }
```

I don't really understand what the Gh60 and your copied code is doing...

Dbroqua, 2017-05-30:
Hello,

With your modification that's works ! Thanks

The GH60 code allows to on/off some leds by pressing key for example.

In the GH60 we have some function leds:
- Capslock
- Fn 
- WASD
- Esc

Each block of leds can be turned on/off separately with this functions.

Regards,
Damien

closed 2017-05-31 by Dbroqua

drc0, 2018-12-08:
hoping you can hep me for a similar issue, my caps lock doesn't stay on, when I press it it goes on and off immediatedly (the light is visible for less than 200ms). This is the code I'm using:

```
void led_init_ports(void) {
    // * Set our LED pins as output
    DDRD |= (1<<6);
}
void led_set_kb(uint8_t usb_led)
{
    if (usb_led & (1<<USB_LED_CAPS_LOCK)) {
      // Output high.
      PORTD |= (1<<6);
    } else {
      PORTD &= ~(1<<6);
    }
}
```
if for example I remve the else branch the led remains on when I click it the first time. Any hint? I need to add I'm porting some working code I had for an old tmk version and the caps lock led was working perfectly with this code:
```
void led_set(uint8_t usb_led)
{
    if (usb_led & (1<<USB_LED_CAPS_LOCK)) {
        // Output high.
        DDRD |= (1<<6);
        PORTD |= (1<<6);
    } else {
        // Output low.
        DDRD &= ~(1<<6);
        PORTD &= ~(1<<6);
    }
}

```