qmk/qmk_firmware #1352: How to map Command + Esc to send Command + Grave?

johndierks, 2017-05-31:
I'm working on a TADA/SABER and have ESC on the number row where grave usually sits.

When I hit Command + Esc, I'd like the kb to send Command + Grave, and Command + Fn + Esc to send Command + Esc.

Otherwise I want Command and Esc to work as they already do.

nikchi, 2017-05-31:
The functionality you're looking for might be in `/quantum/process_keycode/process_combo.c`.

There's no official documentation for that feature as of right now, but examples are available in #960

tmk, 2017-05-31:
You can use `ACTION_LAYER_MODS()` for the Command.

This is for Shift but you can still get the idea from there.
https://github.com/tmk/tmk_keyboard/blob/master/keyboard/fc660c/unimap_emu.c#L9-L10

skullydazed, 2017-06-24:
Now that #1391 has been merged you can use `KC_GESC` for this functionality.

closed 2017-06-27 by skullydazed