qmk/qmk_firmware #1360: PAM2804 controler

Dbroqua, 2017-06-03:
Hello all !

I'm trying to add backlit for my personal keyboard. After some searches and discussion with other maker a good solution is to use PAM2804 controler.

QMK is able to manage this controler ?

Sorry if the question is noob but I'm not a specialist with electronic / firmware :D 

Regards,
Damien

fredizzimo, 2017-06-03:
I don't know much about hardware either, but reading the datasheet, it seems like a good way to make sure that the LEDs don't draw too much current from the USB port.

But as far as the software goes, yes QMK can easilly handle it, and drive the EN pin through a PWM to support dimming and other effects. 

With ChibiOS and ARM processors, there's also support for [IS31FL3731C](http://www.issi.com/WW/pdf/31FL3731C.pdf), which can control each LED individually. But that's much more complex to wire up, and the ARM processor support is still quite untested, so it's probably out of scope for you.

Dbroqua, 2017-06-03:
Hello,

Thanks for your reply, you confirm my investigations.

Regards,
Damien

closed 2017-06-24 by skullydazed