qmk/qmk_firmware #1364: ps2avrGB Make errors

digisax, 2017-06-05:
I'm getting errors when trying to make the hex file, I'm sure it's just something I'm doing wrong but I wanted to post to find out how to fix it. 

The errors are all in matrix.c and it's saying the ports, pins, and ddrs are undeclared. 

I also have a warning for "device type not defined" 

In case it matters I'm using the Ubuntu/Bash subsystem for Windows in powershell.

Thanks
[image]

fredizzimo, 2017-06-05:
The gcc-avr package in Ubuntu 14.04LTS is too old for that controller. 

The easiest thing to do is to update to 16.04LTS, following our [instructions](https://docs.qmk.fm/Build-Environment-Setup.html)!

It is also possible to compile a new version of gcc-avr from the sources, I did that a couple of months ago, but it was far from straightforward and took a long time. The ready compiled binaries did not work in WSL for some reason.

digisax, 2017-06-05:
Thanks for the help, that got the pin, port, and ddr issues fixed, now though it's saying 'macro parameters must be comma separated' in the ps2avrGB.c

[image]

luizribeiro, 2017-06-06:
Pasting your keymap file or patch is probably the best way to get help on this, but I think you're just missing a comma in your macro parameters :)

digisax, 2017-06-06:
The kyemap is pretty simple right now, I just wanted to get it working first
`/*
Copyright 2017 Luiz Ribeiro <luizribeiro@gmail.com>

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, either version 2 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

You should have received a copy of the GNU General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#include "ps2avrGB.h"

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
    KEYMAP( \
        ESC, F1,  F2,  F3,  F4,  F5,  F6,  F7,  F8,  F9,  F10, F11, F12, PSCR,HOME,END, KC_CALC, KC_MUTE, KC_VOLD, KC, VOLU, \
        GRV, 1,   2,   3,   4,   5,   6,   7,   8,   9,   0,   MINS,EQL, BSPC,     DEL, KC_NLCK, KC_PSLS, KC_PAST, KC_PMNS, \
        TAB, Q,   W,   E,   R,   T,   Y,   U,   I,   O,   P,   LBRC,RBRC,BSLS,     INS, KC_P7, KC_P8, KC_P9, KC_PPLS, \
        CAPS,A,   S,   D,   F,   G,   H,   J,   K,   L,   SCLN,QUOT,ENT,           PGUP,KC_P4, KC_P5, KC_P6, \
        LSFT,Z,   X,   C,   V,   B,   N,   M,   COMM,DOT, SLSH,RSFT,          UP,  PGDN, KC_P1, KC_P2, KC_P3, KC_PENT\
        LCTL,LALT,LGUI,          SPC,                     RGUI,RALT,RCTL,LEFT,DOWN,RGHT, KC_P0, KC_PDOT\
    ),
};

const uint16_t PROGMEM fn_actions[] = {
};
`

What I mostly changed was setting up the matrix for my board in ps2avrGB.h
`/*
Copyright 2017 Luiz Ribeiro <luizribeiro@gmail.com>

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, either version 2 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

You should have received a copy of the GNU General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifndef KEYMAP_COMMON_H
#define KEYMAP_COMMON_H

#include "keycode.h"
#include "action.h"

#define KEYMAP( \
  K05, K25, K35, K45, K55, K06, KA6, KA7, K07, KB5, KC5, KD5, KE5, KD1, KE1, KE2, K65, K75, K85, K95, \
  K04, K14, K24, K34, K44, K54, K16, KB6, KB7, K17, KA4, KB4, KC4, KE4,      KD0, K64, K74, K84, K94, \
  K03, K13, K23, K33, K43, K53, K26, KC6, KC7, K27, KA3, KB3, KC3, KD3,      K67, K63, K73, K83,      \
  K02, K12, K22, K32, K42, K52, K36, KD6, KD7, K37, KA2, KB2, KD2,           KE0, K62, K72, K82, K92, \
  K01, K11, K21, K31, K41, K51, K46, KE6, KE7, K47, KA1, KB1,           K86, K77, K61, K71, K81       \
  K00, K10, K20,           K56,                     K57, KB0, KC0, K66, K76, K96, K60,      K80, K90  \
){ \
  { K00, K10,   K20, K30, KC_NO, KC_NO,   K60,   KC_NO, K80,   K90,   KA0,   KB0, KC0,   KD0,   KE0   }, \
  { K01, K11,   K21, K31,   K41,   K51,   K61,   K71,   K81,   KC_NO, KA1,   KB1, KC1,   KD1,   KE1   }, \
  { K02, K12,   K22, K32,   K42,   K52,   K62,   K72,   K82,   K92,   KA2,   KB2, KC2,   KD2,   KE2   }, \
  { K03, K13,   K23, K33,   K43,   K53,   K63,   K73,   K83,   KC_NO, KA3,   KB3, KC3,   KD3,   KC_NO }, \
  { K04, K14,   K24, K34,   K44,   K54,   K64,   K74,   K84,   K94,   KA4,   KB4, KC4,   KC4,   KE4   }, \
  { K05, KC_NO, K25, K35,   K45,   K55,   K65,   K75,   K85,   K95,   KC_NO, KB5, KC5,   KD5,   KE5   }, \
  { K06, K16,   K26, K36,   K46,   K56,   K66,   K76,   K86,   K96,   KA6,   KB6, KC6,   KD6,   KE6   }, \
  { K07, K17,   K27, K37,   K47,   K57,   K67,   K77,   K87,   K97,   KA7,   KB7, KC7,   KD7,   KE7   } \
} 

#define KC_KEYMAP( \
    K05, K25, K35, K45, K55, K06, KA6, KA7, K07, KB5, KC5, KD5, KE5, KD1, KE1, KE2, K65, K75, K85, K95, \
    K04, K14, K24, K34, K44, K54, K16, KB6, KB7, K17, KA4, KB4, KC4, KE4,      KD0, K64, K74, K84, K94, \
    K03, K13, K23, K33, K43, K53, K26, KC6, KC7, K27, KA3, KB3, KC3, KD3,      K67, K63, K73, K83,      \
    K02, K12, K22, K32, K42, K52, K36, KD6, KD7, K37, KA2, KB2, KD2,           KE0, K62, K72, K82, K92, \
    K01, K11, K21, K31, K41, K51, K46, KE6, KE7, K47, KA1, KB1,           K86, K77, K61, K71, K81       \
    K00, K10, K20,           K56,                     K57, KB0, KC0, K66, K76, K96  K60,      K80, K90  \
) \
{ \
    { KC_##K00, KC_##K10, KC_##K20, KC_##K30, KC_NO,    KC_NO,    KC_##K60, KC_NO,    KC_##K80, KC_##K90, KC_##KA0, KC_##KB0, KC_##KC0, KC_##KD0, KC_##KE0 }, \
    { KC_##K01, KC_##K11, KC_##K21, KC_##K31, KC_##K41, KC_##K51, KC_##K61, KC_##K71, KC_##K81, KC_NO,    KC_##KA1, KC_##KB1, KC_##KC1, KC_##KD1, KC_##KE1 }, \
    { KC_##K02, KC_##K12, KC_##K22, KC_##K32, KC_##K42, KC_##K52, KC_##K62, KC_##K72, KC_##K82, KC_##K92, KC_##KA2, KC_##KB2, KC_##KC2, KC_##KD2, KC_##KE2 }, \
    { KC_##K03, KC_##K13, KC_##K23, KC_##K33, KC_##K43, KC_##K53, KC_##K63, KC_##K73, KC_##K83, KC_NO,    KC_##KA3, KC_##KB3, KC_##KC3, KC_##KD3, KC_NO    }, \
    { KC_##K04, KC_##K14, KC_##K24, KC_##K34, KC_##K44, KC_##K54, KC_##K64, KC_##K74, KC_##K84, KC_##K94, KC_##KA4, KC_##KB4, KC_##KC4, KC_##KD4, KC_##KE4 }, \
    { KC_##K05, KC_NO,    KC_##K25, KC_##K35, KC_##K45, KC_##K55, KC_##K65, KC_##K75, KC_##K85, KC_##K95, KC_NO,    KC_##KB5, KC_##KC5, KC_##KD5, KC_##KE5 }, \
    { KC_##K06, KC_##K16, KC_##K26, KC_##K36, KC_##K46, KC_##K56, KC_##K66, KC_##K76, KC_##K86, KC_##K96, KC_##KA6, KC_##KB6, KC_##KC6, KC_##KD6, KC_##KE6 }, \
    { KC_##K07, KC_##K17, KC_##K27, KC_##K37, KC_##K47, KC_##K57, KC_##K67, KC_##K77, KC_##K87, KC_##K97, KC_##KA7, KC_##KB7, KC_##KC7, KC_##KD7, KC_##KE7 }  \
}

#endif
`

fredizzimo, 2017-06-10:
The message does indeed indicate there's a missing comma somewhere. I didn't spot it when checking quickly though.

There could also be some whitespace issue, you can toggle display of whitespace on in your editor to see if that's the case.

jetpacktuxedo, 2017-06-14:
There is a missing comma after K81 (at the end of the line) on the line before both of those error messages. Add a comma to the end of lines 29 and 47 and see if that fixes you up.

rask, 2018-03-10:
I'm getting similar errors on `make pearl:default` on 14.04, will have to see if compiling a newer version of `gcc-avr` helps with this.

drashna, 2018-03-25:
Looks like this is resolved.

If not, then reopen or open a new issue.

closed 2018-03-25 by drashna

rask, 2018-03-26:
Yeah, this one needs `gcc-avr` 4.9, Ubuntu 14.04 for instance has 4.8.

drashna, 2018-03-26:
For somebody not versed in linux, how would one go about updating gcc-avr?
(in part, because I'd like to add this to the faq in the docs)

rask, 2018-03-26:
I am not an expert and I could not find canonical sources for `gcc-avr` so I just gave up and upgraded to Ubuntu 17.10. :D

drashna, 2018-03-26:
hahaha, okay, noted for the future at least. :P

fredizzimo, 2018-04-07:
Another way is to install it manually from here http://www.microchip.com/avr-support/avr-and-arm-toolchains-(c-compilers)

For a 64-bit linux `AVR 8-bit Toolchain 3.5.4 - Linux 64-bit`, that comes with gcc 4.9.2