qmk/qmk_firmware #1365: One column not working

galengold, 2017-06-05:
[My code](https://github.com/galengold/split70/tree/master/qmk_firmware/keyboards/split70)

I'm building a keyboard that's like the ergodox (split, similar shape [layout](http://www.keyboard-layout-editor.com/#/gists/3709167c21216ea1294874d039502740)).

When I flash my modified QMK, everything works except for column 6, attached to Teensy port B0 (left side of the right hand). I've checked and made sure it's not a PCB problem or a construction problem by doing connection tests and checking the wiring. I believe I've modified everything correctly, and going through the code I think everything is right.

I have confirmed that this is purely a code issue by switching the bit shift for "rows" (they're physically columns) 6 and 7 in matrix.c:

```c
        switch (row) {
            case 6:
                DDRB  |= (1<<0);    # becomes (1<<1)
                PORTB &= ~(1<<0);   # becomes ~(1<<1)
                break;
            case 7:
                DDRB  |= (1<<1);    # becomes (1<<0)
                PORTB &= ~(1<<1);   # becomes ~(1<<0)
                break;
```

This confirmed the column and Teensy port worked. U, J, and M were output when Y, H, and N were pressed.

I have no idea why column 6 does this, and I am not familiar enough with the whole project to debug it into the protocol parts etc. Any ideas?

fredizzimo, 2017-06-05:
It appears like you have an one off error in `read_cols`, that check should be `<6`, not `<7`

galengold, 2017-06-05:
That would do it. Thanks for the second pair of eyes.

closed 2017-06-05 by galengold