qmk/qmk_firmware #1367: keycode access in tap dance functions

shieldsd, 2017-06-06:
I have the following tap dance function:
```
void tap_dance_triple(qk_tap_dance_state_t *state, void *user_data) {
    switch(state->count) {
        case 2:
            register_code(KC_LSFT);
            tap(state->keycode);
            unregister_code(KC_LSFT);
            break;
        case 3:
            tap(state->keycode);
            tap(state->keycode);
            break;
        default:
            tap(state->keycode);
            break;
    }
}
```
If the keycode is hard coded e.g. KC_DOT, everything works fine. If I try and access the keycode  from the tap dance state it doesn't appear to be set. Am I missing something?

shieldsd, 2017-06-06:
Just realised the keycode is the tap dance key rather than the target. I'm passing it in via the user data now and it's working.

closed 2017-06-06 by shieldsd

e-carlin, 2019-06-24:
@shieldsd Can you post your solution? I want to do the same thing but I'm not sure how to pass it in via user data. I tried to look through your qmk_firmware fork to see if I could find your changes but was unsuccessful.

shieldsd, 2019-06-24:
The original commit is here:

https://github.com/qmk/qmk_firmware/pull/1368/commits/ad7a5bdc4f2527f47ca22885e5d0b689fb8d4518