qmk/qmk_firmware #1369: Caps lock LED dimming

alwynster, 2017-06-06:
Am I understanding the code correctly that the Caps lock LED (on a Satan GH60) board should be dimmable with the rest of the backlight. Although the backlight is directly connected to the PWM, should the Caps LED also be manageable from that ISR? 

Right now the only things that affects that LED is if Caps lock is turned on, otherwise it remains off (or remains on if  I hard code that in the led_set_kb function)

jackhumbert, 2017-06-06:
Yeah, the soft-PWM backlight functionality can be used in that way, if you assign the backlight pin to your caps lock LED. I'm not sure it can be used in conjunction with the actual backlight though - that may require some additional code.

alwynster, 2017-06-08:
If you're interested, ended up fixing it with the following addition to my satan.c file: 

```
#ifdef BACKLIGHT_CAPS
  uint8_t backlight_on;
  ISR(TIMER1_COMPB_vect)
  {
      led_set_kb(~(1 << USB_LED_BL));
  }

  ISR(TIMER1_OVF_vect)
  {
        if (OCR1B != 0)
            led_set_kb(1 << USB_LED_BL);
  }
#endif

void led_init_ports(void) {
    // * Set our LED pins as output
    DDRB |= (1<<2);
#ifdef BACKLIGHT_CAPS
    // register the required ISRs
    TIMSK1 |= _BV(OCIE1B) | _BV(TOIE1);
#endif
}

void led_set_kb(uint8_t usb_led) {
    if ((usb_led & (1 << USB_LED_BL)) || (layer_state == 2)) {
        // Turn capslock on
        PORTB &= ~(1<<2);
    } else {
        // Turn capslock off
        PORTB |= (1<<2);
    }
}
```

and then to satan.h:

```
#define BACKLIGHT_CAPS
#define USB_LED_BL 5 // alternative to Caps LED set by USB protocol
```

In short, my "Caps lock" button is mapped to FN. This means the Caps LED follows the brightness of the backlight and stays bright while I press any FN key.

Thought I'd post it in case it bothers someone else too

iFreilicht, 2017-06-27:
@skullydazed I think you can close this :)

closed 2017-06-27 by fredizzimo