qmk/qmk_firmware #1370: weird extra keycode being send if release LT() before another key

dunkarooftop, 2017-06-07:
I downloaded the latest version few days ago and recompile the keyboard and found this issues, not sure if it's something wrong on my end or it's a new bug. say

Base layer: character "A" lay on top of Number layer: "1"
When I pressed the key to enter Number layer (let's call it Numb Key here) 

If I press down Numb Key, then tap "A" then release Numb Key will send "1" which is what we want.
But 
if I press down Numb, -> press down "A", -> let go of Numb Key, -> let go "A" key will send "1", "a" instead of just "1"

I don't think it used to behave this way  can someone try it and see if this happen to too thanks :)

dunkarooftop, 2017-06-07:
turn out it's cause by enabling the ```combo```, the problem go away after I disable it. The ```combo``` function is still a bit buggy #960

fredizzimo, 2017-06-10:
I don't know what's wrong, but I can check this more carefully at some point.

I think many things are getting quite complex, especially in how different features interact with each other, in this case the tapping and combo, so I really think we need to get some unit tests added, to both help with the implementation, and to make sure that nothing breaks.

fredizzimo, 2017-06-10:
I think I found the issue, and commented on the pull request. But because it's not easy to fix I suggest that you don't use the same keys for LT and combos.

dunkarooftop, 2017-06-11:
@fredizzimo 
Thank you for looking into it and thank all the team who work hard for the community : )

drashna, 2018-03-25:
Sounds like this is resolved? 

If not, then please reopen or open a new issue.

closed 2018-03-25 by drashna