qmk/qmk_firmware #1399: WS2812 support for ARM based keyboards

paxy97, 2017-06-15:
It would be nice to add RGB underglow to keyboards that use ARM controllers like the infinity ergodox.

There is the OctoWS2812 library for the Teensy which has the same MCU as the Ergodox, so it might be porable to QMK.

https://github.com/PaulStoffregen/OctoWS2811

drashna, 2018-03-25:
This would be nice, all things considered.... 
@jackhumbert (or whomever I should tag)

chax, 2018-11-08:
@jackhumbert i saw you made some progress regarding this feature on branch [arm_rgb](/qmk/qmk_firmware/tree/arm_rgb)  and i tried to play with it a little on my rev6 planck.
I also found few reddit posts where it was discussed about testing this branch:
https://www.reddit.com/r/olkb/comments/9kkxuu/getting_the_rgb_underglow_working/
https://www.reddit.com/r/olkb/comments/9jrfzb/planck_leds_reversed/
https://www.reddit.com/r/olkb/comments/9uzci9/rev6_with_leds_help/

I found out that timers don't work on this branch, therefore some features won't work also (like `LT(layer, kc)`) and sometimes song won't play on RESET and so on...

i found fix for this issue and made it work for me. I described solution in this comment:
https://www.reddit.com/r/olkb/comments/9jrfzb/planck_leds_reversed/e95eb6g/

I also noticed that this branch is a bit outdated, and as i tried to merge master to it locally (and succeeded) there were a lot of conflicts.

I hope I helped a bit to get this in master branch a little bit sooner because people that got their rev6 want to play with all available features. I'm currently running my local custom branch so it works for me, but more and more people are asking for this feature.

jackhumbert, 2020-03-13:
We have this implemented now :) [The docs for it are here](https://docs.qmk.fm/#/ws2812_driver).

closed 2020-03-13 by jackhumbert