qmk/qmk_firmware #1404: CTL_T(LCTL(KC_BSPC)) only outputs backspace on tap

Krakob, 2017-06-17:
Using an XD60, with 

`CTL_T(KC_BSPC)` on Caps Lock, layer 0.
`LT(1, KC_SPC)` on space, layer 0.

`CTL_T(LCTL(KC_BSPC))` on Caps Lock, layer 1.
`KC_TRNS` on space, layer 1.

The idea being that caps is Ctrl when held, backspace when pressed. Since I can't do both at once, I'd like space+caps output Ctrl+backspace (deletes a full word backwards), but it's just giving me backspace :(

Everything else with the space Fn works just fine.

jackhumbert, 2017-06-18:
Unfortunately [you can't use those keycodes in `CLT_T` like that](https://docs.qmk.fm/key_functions.html#limits-of-these-aliases) :/

Krakob, 2017-06-18:
Aw jeez. Any suggestion on a workaround? It was to me that I should use macros but I'm not sure exactly how I'd do that. I guess I'd have to re-implement the hold/tap inside of one?

h-youhei, 2017-06-20:
How about #1046 ?
In the PR, File changed, keymap.c will help you.

In this case, use `MACRO_TAP_HOLD_MOD`;
pass this to macro parameter:
  `MACRO(D(LCTL), T(BSPC), U(LCTL), END)`
pass `LCTL` to mod parameter.

Sorry I haven't tested.

h-youhei, 2017-06-20:
[Here](https://docs.qmk.fm/macros.html) is macro documentation.

Krakob, 2017-06-30:
Finally had the time to test it today. Using

```c
const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt) {
  switch(id) {
    case 0:
      if(record->event.pressed) {
        return MACRO_TAP_HOLD_MOD(record, MACRO(D(LCTL), T(BSPC), U(LCTL), END), LCTL);
      }
      break;
  }

  return MACRO_NONE;
}
```

And I'm getting ctrl upon holding it, but the tap does nothing.

h-youhei, 2017-07-01:
https://gist.github.com/h-youhei/c77d626a097799c1c6345d4cc0fa08b8

This code works like you described.
I tested with Ergodox EZ.

Krakob, 2017-07-01:
It works, it works! Thank you a bunch! <3

closed 2017-07-01 by Krakob