qmk/qmk_firmware #1411: Combos don't play well with leader keys

mneme, 2017-06-19:
When shadowing combos and leader keys the shadowed keys get sent by the keyboard even if you are in a leader combination.

In my case I have a combo for `asdf` and a leader string using `s => s`

Expected:
`LEAD => s  => s = ¯\(ツ)/¯`

Result: 
`LEAD => s => s = ss¯\(ツ)/¯`

Probably a pretty hard interaction to actually fix but I just thought I'd throw it up here anyways.

drashna, 2018-03-25:
Hmmm, that's pretty nasty though.

Talljoe, 2018-04-22:
`process_leader` [allows the keyup to go through](https://github.com/qmk/qmk_firmware/blob/master/quantum/process_keycode/process_leader.c#L36). This causes `process_combo` to think a key has been released and [sends it out](https://github.com/qmk/qmk_firmware/blob/master/quantum/process_keycode/process_combo.c#L91-L103).

AnonJohn, 2020-02-10:
FWIW, this no longer seems to be an issue. Fixed?

jackhumbert, 2020-03-13:
If that's not the case, feel free to comment/reopen - thanks!

closed 2020-03-13 by jackhumbert