qmk/qmk_firmware #1415: Ergodox Infinity backlight doesn't work reliably

fredizzimo, 2017-06-21:
Continuing the discussion from #1361.

By @packman86

> The backlighting works but controlling it seems a bit buggy with the BL_ON, INC, DEC, TOGG, STEP. On mine, it I can not toggle it on or off. I can only toggle between the 3 different brightnesses, and the 4th step is off.

My Response
> Could you please give more details. 
Does `BL_TOGG`, `BL_ON`, and `BL_OFF` all do nothing? I believe at least `BL_TOGG` was working for @333fred, so I need to know if there are some additional steps that you are doing, which makes it not work.
>
>When you turn the backlight off and then on again it should return to the same brightness as it had before. Please note that brightness 0 is special, it both turns the backlight off and sets the brightness to 0 zero, so when you try to turn it on from that setting, nothing will actually appear to happen. You need to increase the level for it to turn on.
>
>That is the standard QMK behaviour, which might not make complete sense, better would probably be to always force at least level 1. But I don't want to change that, at least not now, without discussing it with a broader audience first.

And another response from @packman86
> @fredizzimo The various BL_ seem to work sometimes. I have issues where sometimes the BL_ commands only work on one hand. And sometimes it triggers a small cluster of LEDs to turn off or on.
I can't find a pattern for when this happens or what causes these glitches. Restarting seems to help. I know the backlighting works on my Ergodox since it is fine on the normal Input Club firmware.

Here's my response to that:
That sounds very odd. Does the glitches only happen when you change change the backlight, or does it happens sometimes, when you just leave the backlight on for an extended period of time?

Does anyone else experience similar issues?

333fred, 2017-06-21:
Toggle, inc, and dec all seem to work correctly for me.

packman86, 2017-06-21:
It happens when I press any of the BL_ buttons. I am going to try and reinstall the keymaps when I get home tonight.

packman86, 2017-06-23:
https://streamable.com/trn9z

That is a video showing how my keyboard behaves. The BL_TOGG seems to be the only button that behaves as it should. BL_ON and BL_OFF seem to be doing the same thing. And the right hand doesn't light up at all.

fredizzimo, 2017-06-23:
For the right half, are you sure that you have flashed it separately with the correct code?

`BL_ON` and `BL_OFF` calls `backlight_level` with 9 respective 0. No other keymap seems to be using those, and the function is totally buggy, can you spot the error?

```C
void backlight_level(uint8_t level)
{
    backlight_config.level ^= level;
    backlight_config.enable = !!backlight_config.level;
    eeconfig_update_backlight(backlight_config.raw);
    backlight_set(backlight_config.level);
}
```

The function also doesn't make sure that the backlight level is within the correct range. I will fix this within the next few days.

packman86, 2017-06-23:
My build command is:

make ergodox-infinity-doxxy VISUALIZER_ENABLE=yes LCD_BACKLIGHT_ENABLE=yes LCD_ENABLE=yes BACKLIGHT_ENABLE=yes MASTER=left

and of course MASTER=right for the right hand file.

fredizzimo, 2017-06-23:
But did you connect the right hand to the computer alone and flash it? Both halves need to be flashed separately.

packman86, 2017-06-25:
Yes that is what I did. I unplug whichever hand is not being flashed. I connect the hands into the computer via USB hub in my monitor.

packman86, 2017-06-26:
I have tried adding "BL_" commands on both hands but still only the left hand works.
As mentioned in other thread, both hands work fine when using the I:C firmware. https://i.imgur.com/mP3QWeT.jpg

fredizzimo, 2017-07-08:
Does anyone else experience this? Any additional information or help with debugging this issue is would greatly be appreciated.

I found an uninitialized variable related to the backlight LEDs. I don't think it can explain all the issues here, but I will send a fix for it quite soon.

fredizzimo, 2017-07-08:
I opened a pull request #1475, which fixes the unititialized variable, could you please check if that fixes the problem?

r2d2rogers, 2017-07-13:
```BL_TOGG```, ```BL_INC```, and ```BL_DEC``` all seem to work correctly for me, but ```BL_STEP``` causes the lowest level of back light to flash to full brightness momentarily before assuming the correct level.  ```BL_/INC/BL_DEC``` work without this behavior and ```BL_TOGG`` turns off the back light and turns it back on the the correct level it was at before the toggle. The flash is present on both the master and the slave simultaneously. 

I can do video to document, but I have not done so yet due to the simpler behavior versus what I reported in gitter and corrected.

tenderlove, 2017-09-29:
I'm seeing similar behavior to what @packman86 described.  Only the left side will light up (I've only tried `BL_INC` and `BL_DEC` though).  What can I do to help debug?

fredizzimo, 2017-09-29:
Good timing, we just found an issue with @r2d2rogers. I will try to send a pull request tomorrow, but I'm not entirely sure it's the same issue.

tenderlove, 2017-09-29:
@fredizzimo well I can at least test the PR. 😊

fredizzimo, 2017-09-30:
The pull request has now been submtted. https://github.com/qmk/qmk_firmware/pull/1776

dsvensson, 2017-10-25:
@jackhumbert PR got merged, resolved?

jackhumbert, 2017-11-03:
Yeap! Thank :)

closed 2017-11-03 by jackhumbert