qmk/qmk_firmware #1428: How to Configure a Second Backlight?

axtran12, 2017-06-25:
Looks like BL commands are limited to just the main backlight. For the NerD PCBs, there's a second backlight that's more commonly known as the Side LEDs--is there a simple way to define a second set of backlight functionality as BL2_TOGG, BL2_DEC, BL2_INC which respects the existing control functionalities?

jackhumbert, 2017-06-25:
This is something I'd like to see support for! Are they single color LEDs, or RGB ones?

axtran12, 2017-06-25:
For the GON PCBs, it's a single color. The functions call it out as Backlight_PCB above, however they do reference an older TMK which I believe did not support level control in the lighting API.

jackhumbert, 2017-06-25:
We were chatting about this a bit on Gitter, and thought it might be cool to develop an array of lighting implementations (similar to how the keymap is defined) supporting all backlights, rgb lights, keyswitch lights, etc, and control them with keycodes/functions dedicated to each index, or via an argument passed through. 

It'd be possible to have any number of implementations enabled for a keyboard, and use/control them all individually (or together). This could start with a basic high/low implementation for status leds and simple backlights, and work our way up to supporting more advanced things like addressable and RGB LEDs.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc