qmk/qmk_firmware #1430: Command keys of S60-x suddenly don't work on Macbook

wyudong, 2017-06-25:
My keyboard is S60-x and it used to behave greatly with Macbook. But until last week, the left and right command keys don't work when pressed. I tried to [test the keyboard] (http://www.keyboardtester.com/) and two command keys couldn't be recognized by the tester. Then I erased and flashed the firmware but unluckily I got the same results. So I suspected it was my Macbook's fault? But connected the keyboard with other Macbooks, the command keys still don't send any feedback. ALL the others keys works normally.

Can anyone tell the reason for my issue. Thanks for the help.

Dbroqua, 2017-06-25:
Hello,

What layout are you using?

If you use another key for left/right command is it ok? And if you replace left/right command with other code key (like Q) it works?

Have you test the switches?

wyudong, 2017-06-25:
Hi @Dbroqua 

Thanks for your quick reply. I'm using my customized layout. I cloned this repo again did the following tests:

1. Use Q/W for left/right commands. Result: Q/W will send nothing.
2. Use left/right commands for Q/W. Results: commands will send Q/W.

Here is my keymap.c. I put it in *keyboards/s60_x/keymaps/my-keymap*, compile and flash.
 ```c
#include "s60_x.h"

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {

    [0] = KEYMAP(KC_ESC,   KC_1,    KC_2,    KC_3,    KC_4,    KC_5,    KC_6,    KC_7,    KC_8,    KC_9,    KC_0, KC_MINS,  KC_EQL,   KC_NO, KC_BSPC, \
                 KC_TAB,   KC_Q,    KC_W,    KC_E,    KC_R,    KC_T,    KC_Y,    KC_U,    KC_I,    KC_O,    KC_P, KC_LBRC, KC_RBRC, KC_BSLS,          \
                 KC_LCTL,  KC_A,    KC_S,    KC_D,    KC_F,    KC_G,    KC_H,    KC_J,    KC_K,    KC_L, KC_SCLN, KC_QUOT,   KC_NO, KC_ENT ,          \
                 KC_LSFT,  KC_NO,   KC_Z,    KC_X,    KC_C,    KC_V,    KC_B,    KC_N,    KC_M, KC_COMM,  KC_DOT, KC_SLSH,   KC_NO, KC_UP  ,   KC_NO, \
                 KC_FN0, KC_LALT, KC_LGUI,                   KC_SPC,                   KC_RGUI, KC_LEFT, KC_DOWN, KC_RGHT),

    [1] = KEYMAP(KC_GRV,    KC_F1,   KC_F2,   KC_F3,   KC_F4,   KC_F5,   KC_F6,   KC_F7,   KC_F8,   KC_F9,  KC_F10,  KC_F11,  KC_F12, KC_TRNS,  KC_DEL, \
                 KC_TRNS, KC_FN10, KC_FN11, KC_FN12, KC_FN13, KC_TRNS, KC_TRNS, KC_PGUP,   KC_UP, KC_PGDN, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS,          \
                 KC_CAPS, KC_MUTE, KC_VOLD, KC_VOLU, KC_TRNS, KC_TRNS, KC_TRNS, KC_LEFT, KC_DOWN, KC_RGHT, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS,          \
                 KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, \
                 KC_TRNS, KC_TRNS, KC_TRNS,                   KC_TRNS,                   KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS),
};
```

wyudong, 2017-06-25:
Correct me if I'm wrong, S60-x has no switches, only a button.

Dbroqua, 2017-06-25:
S60-X is Sentraq S60-X?

This: https://www.massdrop.com/buy/sentraq-60-diy-keyboard-kit

I said switches for MX/Gateron switches.

wyudong, 2017-06-25:
@Dbroqua The one I bought is https://www.massdrop.com/buy/19526. Since command keys can send signals of Q/W, I believe the switches and board are not damaged.

Dbroqua, 2017-06-25:
Ok,

I have the RGB version of this PCB.

You wrote this:

> Use Q/W for left/right commands. Result: Q/W will send nothing.

When you wrote:
> One more thing found: all keys on row 1 don't work.
You want says for example, SPACE does not works to ?

If it's that I think your PCB are damaged but it's really strange.

wyudong, 2017-06-25:
The row 1 I wrote is `ESC|1|2|3|4|5|6|7|8|9|0|-|=`. The key on this row won't send anything even I map them to other keys, e.g. 1 for Q.

If the PCB is damaged, then it's suicided haha.

Dbroqua, 2017-06-25:
Row 1 = dead
Switch under Command = dead

It's like a suicide!

In mine it's the RGB underglow who died suddenly :/ .

You can contact Brandon from Sentraq: https://sentraq.com/

Regards

wyudong, 2017-06-25:
Thanks for the help @Dbroqua

closed 2017-06-25 by wyudong