qmk/qmk_firmware #1434: Setting MOUSEKEY_WHEEL_DELAY in config.h does nothing

BeatVids, 2017-06-25:
Used to use an old tweak with mousekey.c :
```
void mousekey_task(void) {
    if (!mouse_report.x && !mouse_report.y && !mouse_report.v && !mouse_report.h)
        return;
    uint16_t last_report_age = timer_elapsed(last_timer);
    if (mousekey_repeat) {
        if (last_report_age < mk_interval) return;
    }
    else {
        if (mouse_report.x || mouse_report.y) {
            if (last_report_age < mk_delay * 10) return;
        }
        else if (mouse_report.v || mouse_report.h) {
            if (last_report_age < mk_wheel_delay * 10) return;
        }
    }
```
However, since mousekey.c is a bit different, I'm not sure what tweaks I need to do. I don't think mk_wheel_delay is being used anywhere, and I would like to, but I'm not sure how..

drashna, 2018-03-25:
Yeah, I can't find this used anywhere (where it would matter).

drashna, 2018-10-21:
Is this still an issue?

PR #6685 (2020-03-13): Decouple mouse cursor and mouse wheel in accelerated mode
## Description

In accelerated mode, MOUSEKEY_INTERVAL is used for both the cursor and the wheel. Since the step for the wheel is way higher than the step for the cursor, this results in the wheel being almost unusable. See issue #3370 for example.

In this change, I decoupled the cursor and the wheel behaviors (using the decoupling pattern in the constant mode). I also introduced 2 settings:

- MOUSEKEY_WHEEL_DELAY
- MOUSEKEY_WHEEL_INTERVAL

which are the equivalent of MOUSEKEY_DELAY and MOUSEKEY_INTERVAL. Users can therefore choose di [... cut]
closed 2020-03-13 by jackhumbert