qmk/qmk_firmware #1437: Odd behavior on my Ergodox Infinity

packman86, 2017-06-26:
I am getting some really odd behavior with my Ergodox Infinity. The backlighting only seems to want to work on my left hand. When I make changes to my keymap for the right hand, remnants of the keymaps seem to stick around.

Here is my

fredizzimo, 2017-06-26:
Exactly what do  you mean by?
> remnants of the keymaps seem to stick around

The keymap is read purely from the keyboard that is plugged in to the computer, the other half just send the raw keypresses. So if you try to flash the right hand only then no changes will be visible.

For the LED issue, I would rather track that in #1415, but have you tried connecting the other half to the computer, does the LEDs light up in that case? And is the other half dark? 

If it's always the same half that is dead, no matter which way you have connected it to the computer, the problem is most likely a hardware issue, rather than a software one. Have you tried with the official Input Club firmware?

packman86, 2017-06-26:
I make sure to flash both hands every time I make changes to the keymap file.

I tested the I:C firmware last night and both hands lit up perfectly fine.
https://i.imgur.com/mP3QWeT.jpg (continuing backlighting talk in #1415 )

drashna, 2018-03-25:
Does this still happen?

closed 2018-10-21 by drashna