qmk/qmk_firmware #1438: How do you compile QMK for ergodox infinity?

ExperimentsWithCode, 2017-06-27:
I cloned the main qmk repo with `git clone...`

I added the chibios stuff with `git submodule update --init --recursive`

I wrote a keymap. Though I am not clear on if I should put it in which of the following:

- /ergodox/keymaps/MYFOLDER/keymap.c
- /ergodox/infinity/keymaps/MYFOLDER/keymap.c
- ergodox/infinity/keymap.c

From there I navigate into the ergodox folder and have tried:

- make clean
- make inifinity-keymap
- make keymap

All of them give me this warning: 
`/bin/sh: arm-none-eabi-gcc: command not found`

skullydazed, 2017-06-27:
You need to tap the `gcc-arm-embedded` cask, and then you'll be able to install `gcc-arm-none-eabi`. For flashing you will want to also install `dfu-util`. Setting up the ChibiOS build environment is something we need to better document.

As far as the keymap stuff goes, the `ergodox/infinity/keymaps` directory is for keymaps that only work on the ergodox infinity. If your keymap can work on any ergodox (EZ, classic, or infinity) you should put it into `ergodox/keymaps/MYFOLDER/`.

ExperimentsWithCode, 2017-06-27:
I got 

'Invalid tap name 'gcc-arm-embedded'

I succeeded by doing 

'brew tap PX4/homebrew-px4
brew update
brew install gcc-arm-none-eabi'

I found that in another git thread here [here](https://gist.github.com/joegoggins/7763637)

When I run make keymaps in the ergodox folder, it runs successfully. 

When I run make infinity-keymaps it still runs all the keymaps in ergodox as well as my keymap in infinity but gets this same error on all of them:

'Making ergodox/infinity with keymap experimentswithcode and target keymaps                                    [ERRORS]
make[1]: *** No rule to make target `keymaps'.  Stop.'

skullydazed, 2017-06-27:
Interesting. On my machine I this is what I have:

```
$ brew cask list
gcc-arm-embedded
$ brew search gcc-arm-none-eabi
px4/px4/gcc-arm-none-eabi                   px4/px4/gcc-arm-none-eabi-54
px4/px4/gcc-arm-none-eabi-47                px4/px4/gcc-arm-none-eabi-62
px4/px4/gcc-arm-none-eabi-48                px4/px4/gcc-arm-none-eabi-63
px4/px4/gcc-arm-none-eabi-49 ✔
```

Sorry to have led you astray there.

Try `make infinity-MYFOLDER` and see if that works.

ExperimentsWithCode, 2017-06-27:
Try make infinity-MYFOLDER and see if that works. **< that worked**

brew search gcc-arm-none-eabi **< I get the same results. Weird it wouldnt install**

Thank you.

skullydazed, 2017-06-27:
Anytime, glad it was fairly easy.

closed 2017-06-27 by skullydazed