qmk/qmk_firmware #1439: Firmware Readme: Merge all Ergodox variants at one place

htgoebel, 2017-06-27:
From the reader's perspective it's irritating that
- Ergodox ez is prominently in the "Official" section
- ergodox infinity is burred in the "Community-supported" section
- the original Ergodox is missing at all.

AFAIK all Ergodox variants share the same computer and the firmware is (largly) interchangeable.  Thus the other variant should be listed in the Ergodox section. Or at least some statement like "QMK can also be used for other Ergodoc variants, see the contrib section or simply try it out."

lindhe, 2017-06-27:
I was just about to say: what's the point of `/keyboards/ergodox/ez/keymaps` vs `keyboards/ergodox/keymaps`?

ezuk, 2017-06-27:
That's not really true though. The Infinity uses a different processor, and the EZ really is an official QMK partner. @fredizzimo has done enormous work to bring the Infinity on board and got amazing results.

jceaser, 2017-06-27:
I'm in favor of documenting the different in the read me files and at the top level. Newbies such as my self are easily confused by flash code

lindhe, 2017-06-27:
But is there a point in having EZ separate from regular ErgoDox, if it is an official QMK partner?

ezuk, 2017-06-27:
We're actually going to reformat the readme entirely per a new suggestion, which would hopefully clarify this. Originally I wanted to have two separate top-level dirs, `ergodox_ez` and `ergodox_infinity` but there was too much shared code/keymaps. It still might be worth experimenting with.

The idea is definitely not to confuse anyone. The challenge is that there _is_ a difference in terms of the hardware, as well as in terms of the "official" involvement with QMK by the keyboard's originators, so that should probably be clear (to set expectations correctly).

lindhe, 2017-06-27:
I'm currently refactoring my code for the Swedish layout, so it can be pushed to master. Does that mean I should have it under `/keyboards/ergodox/ez/keymaps`, since I wield the ErgoDox EZ?

ezuk, 2017-06-27:
No, not at all -- should go under `keyboards/ergodox/keymaps`. That EZ directory shouldn't really exist.

ezuk, 2017-06-27:
Fixed that in #1440

fredizzimo, 2017-06-27:
The directories `keyboards/ergodox/ez/keymaps` and `keyboards/ergodox/infinity/keymaps`, are reserved for keymaps that are compatible with only either the Ez or Infinity.

The reason for the `profet80` keymap to be in the Ez folder is because there's no 80 key variants of the Infinity, so it's not compatible. Well there's not really any 80 key variants of the Ez either, but for the Ergodox, the Ez and original Ergodoxes are synonyms.

I guess another way to fix it would be to define the 80 key macro even for the Infinity, and make it compile, but that would still not mean that it works.

This folder structure is not specific to the Ergodox, the other keyboards with subprojects work the same way.

ezuk, 2017-06-27:
Ah, thanks! Reverted that then. :)

drashna, 2018-03-25:
These keyboards were separated out.  No longer an issue.

closed 2018-03-25 by drashna