qmk/qmk_firmware #1451: Cannot get any single Satan GH60 keymap to work

malattia, 2017-06-30:
This is my first programmable keyboard, so bear with me.
I'm on Linux.
```
$ make satan-default
Making satan with keymap default

avr-gcc (GCC) 4.9.2
...
Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  17744	      0	  17744	   4550	_satan_default.hex_

$ sudo dfu-programmer atmega32u4 erase
$ sudo dfu-programmer atmega32u4 flash satan_default.hex
Validating...
17744 bytes used (61.89%)
$ sudo dfu-programmer atmega32u4 reset
$  _+{ _+++{\{\
```

So, the last line is what i get after pressing all keys with the default keymap. Not sure how interesting it is, but with all options set to "no" in the Makefile, I get mostly modifier keys instead. Here's a snippet:
```
    state 0x2000, keycode 130 (keysym 0xff31, Hangul), same_screen YES,
    state 0x2000, keycode 130 (keysym 0xff31, Hangul), same_screen YES,
    state 0x2000, keycode 62 (keysym 0xffe2, Shift_R), same_screen YES,
    state 0x2001, keycode 248 (keysym 0x0, NoSymbol), same_screen YES,
...
```

Now for the puzzling part. Using http://kbfirmware.com/ I get a functional keyboard.
I have tried looking at the diff between the QMK they provide and the latest git HEAD, and while there is not much difference in quantum/ (besides removed copyrights!! sigh), tmk_core/ is vastly different.

I have attached the configuration i got from the website:
[mattiasatanamax.json.gz](https://github.com/qmk/qmk_firmware/files/1113936/mattiasatanamax.json.gz)

Anyway, I am wondering where to start to debug this. Help?

skullydazed, 2017-07-03:
If I had to guess I'd say that your PCB has a different hardware layout from the GH60 in QMK. Unfortunately the term GH60 has become very diluted, with a lot of chinese PCB's being called GH60 even though they have a different pin configuration than the original GH60. You'll have to use the Pins tab on http://kbfirmware.com and compare it against the `config.h` for the satan to see how they differ.

malattia, 2017-07-04:
That's something I looked at already. Pins definitions are the same. 
The wiring that i used on kbfirmware.com essentially what this guy depicted in the fourth picture: http://imgur.com/gallery/91fNc

unxmaal, 2017-08-03:
I'm having this same issue.

I reflashed a board using my 'unxmaal' keymap as listed in the QMK master branch for satan/keyboards/keymaps . 

The Satan board became unusable. 

I tried flashing with default keymap, with different results but still failure (two keys worked but emitted the wrong characters). This is as of today, using latest git pull from master, and ensuring all submodules are up to date.

I was able to recover my board by generating a .hex from http://kbfirmware.com/ .

drashna, 2018-03-25:
Sounds like this may be resolved? 

If not, reopen or open a new issue, please.

closed 2018-03-25 by drashna

vatho, 2018-08-26:
@malattia Did you manage to test if this issue still occurs?

Because I think I'm running into the same problem. Using the generated .hex from https://config.qmk.fm with keyboard set to `satan` and layout to `LAYOUT_60_ansi`- I brick my keyboard.

Using the default config for `GH60 Satan (ANSI)` from http://kbfirmware.com/ I can restore it.

drashna, 2018-08-26:
@vatho you may need the GH60 keyboard, not satan.

vatho, 2018-08-26:
Thanks for the quick replay, @drashna! I think I'd already tried the GH60 and bricked the KB as well, but I'll retry and get back to you. 

In the meantime, this is what I've come up with: I think the GH60 config in QMK is for the A and B revisions. The PCB @malattia was using seems to be the GH60 CHN revision. 

I took a quick picture of the PCB I'm working with - note the C6 which is missing on the GH60 CHN revision.

|PCB|Image|Source|
|---|---|---|
|GH60 rev CHN|[image]|https://imgur.com/gallery/91fNc|
|GH60 rev A/B|[image]|https://github.com/qmk/qmk_firmware/tree/master/keyboards/gh60|
|Satan GH60?|![image](https://lh3.googleusercontent.com/W-xbZf4kQ13q8kDHBbYsSAIhcHcMS98t08YWBk1w-CK66vadD56I1y4dVvzlI4OE-guDFVg0f1BQSswuk6H21TUZuF4l1tNzqUvelog00iXuXRlOpuBCkoDX2sYBrn1zYcGET4ruqciU2N50C9HJd1OXKEjhInC9qv3g_FK-DrSGWDQBiKIBdgnMld9OwcuK_XVAXgTIsUmSII3wLRinftbha-5R2ntXa5GOc7IulZzn9loE92fHLb5Zi7eomEKWyL320vvGM6QdhwsyNaLXar0eonzvj56_9ah8jE_YHvA3EoJx4RZ6g_AoQLPguv7G1yHEn9jgA0VjMhQrlC6JjCc7q30Gh0ZCQn5hHEyNV8fm5zGDEmATz6pt7bQmiVrJdcrzQPZv8Gs52ulkEI0kAoqzfrVxgQG42Fu5xt28LThXJvvHpflfHZUAEIMzxB2F_pPH4YaCFhDIBPX5f2LRZJpRsqMl4hVx4u4Uaq8rC1BGqO4cI-KFevU3NNtg2MxhHEjAhgXJiTJQ-KE4MswRuP1TD-MV0aTXO_oTY-RUyRV1mWfXPhVblRK78NzGKPexvSW94UnpfuYKpAN6XGSLCkMAViaTC3M5b-dvGXg6Jva_liYBjdkmAURLGF1NDaby=w1920-h708-no)|my keyboard|

drashna, 2018-08-26:
Looking at the pin config, then yeah, the satan is the one that you want. It's using the same pin configuration as what's on kbfirmware. 

Not sure why it's bricking your board, though

vatho, 2018-08-26:
Bricking may be the wrong word - I just mean it becomes unusable. :) The biggest issue is that when I flash it with a .hex for the Satan board, the `space`+`b` key combo to put the KB in bootloader mode no longer works, so I have to manually short the RESET pin on the chip.

I've just tried flashing a Satan config again and the issue persists. Most of the keys do nothing, some of them output random, wrong characters.

Is there anything I can do to provide you with extra information to debug this?

Another thing I noticed is that the image [here](https://github.com/qmk/qmk_firmware/tree/master/keyboards/satan) does not have the C6 transistor which you can see on my PCB [image]

Sorry if this all sounds a bit confusing, this is my first time messing with reflashing a KB PCB. :)

vprab, 2019-03-23:
@vatho Sorry to resurrect this thread, but did you ever find a solution to this problem? I'm having the same issue and can only assume it's a bug with the firmware that QMK Configurator generates for the Satan.

vatho, 2019-03-23:
I couldn't find a (fully) working solution, no. I tried to adjust some existing configurations and managed to get _some_ keys working, but I'd have to spend some more time learning the details of QMK (and catching up on my C knowledge).

It's on my to-do list, just not very high.

fauxpark, 2019-03-23:
Hmm... [my PCB](https://kbdfans.cn/collections/60/products/satan-gh60chn-pcb-60-layout) has that C6 cap and the `satan` make target has always worked for me.

fauxpark, 2019-03-23:
I found a supposed matrix layout for the C6-less PCBs: https://imgur.com/gallery/91fNc
But the pinout is exactly the same as what's in the satan config.h.

I would not be surprised if there was another revision of this board with a different matrix, but I'm thinking it might just be a dud :/

fauxpark, 2019-03-23:
@vprab I recreated my keymap in the configurator and had no issues with it, I don't think there is a bug there.

vatho, 2019-03-23:
@fauxpark can you share the keymap and settings you used in the configurator? As well as the keyboard/PCB you are flashing to?

If I flash the following keymap from the configurator (https://config.qmk.fm/#/satan/LAYOUT_60_ansi):
[image]

I get the following outputs when going over each key and pressing them:
|KEY|output|
|---|------|
|`  |      |
|1  |<enter>|
|2  |      |
|3  |<menu>|
|4  |      |
|5  |      |
|6  |      |
|7  |      |
|8  |      |
|9  |      |
|0  |      |
|-  |3     |
|=  |      |
|q  |      |
|w  |      |
|e  |      |
|r  |      |
|t  |g     |
|y  |      |
|u  |\     |
|i  |      |
|o  |      |
|p  |      |
|[  |+     |
|]  |      |
|\  |      |
|a  |      |
|s  |      |
|d  |      |
|f  |      |
|g  |      |
|h  |<menu>|
|j  |3     |
|k  |      |
|l  |      |
|;  |g     |
|'  |8     |
|z  |3     |
|x  |g     |
|c  |8     |
|v  |      |
|b  |<menu>|
|n  |<home>|
|m  |      |
|,  |      |
|.  |      |
|/  |      |

I don't know enough about how the matrix on the PCBs works, but I can see some keys returning (g, 3 and 8) which may be a hint as to what the difference in pathing is?

fauxpark, 2019-03-23:
@vatho - same keyboard & layout (`satan`/`LAYOUT_60_ansi`), and basically the same keymap (I'm highly skeptical that the specific arrangement of keycodes matters). I've also ISP flashed the QMK DFU bootloader, but that's not really of any importance either once it's actually running the firmware.

The toolbox output looks normal to me, other than the large firmware size due to the rules.mk stuff not being configurable yet.

Here's a default keymap from current master, using avr-gcc 8.3 (I believe qmk_compiler uses a much older version, perhaps that's the culprit):
[satan_default.hex.zip](https://github.com/qmk/qmk_firmware/files/2999360/satan_default.hex.zip)

vatho, 2019-03-23:
No luck with that hex either, but thanks for trying/compiling!

So the back of your PCB looks like the third one in the table in my [previous comment](https://github.com/qmk/qmk_firmware/issues/1451#issuecomment-416052725)?

fauxpark, 2019-03-23:
It does indeed, even down to the weird misplaced R31 (which I never noticed before... wow).
If you've got a multimeter to hand you could try tracing the matrix according to the image in the album I posted above.

vprab, 2019-04-01:
@vatho I figured it out - you have to reset the EEPROM before flashing the firmware. Not sure why this isn't needed for the kbfirmware hex, but it works :smiley:

abhishekbalam, 2019-04-19:
I am facing the same issue with my GH60 ([lt is the 2nd one from the pic](https://github.com/qmk/qmk_firmware/issues/1451#issuecomment-416052725)).

For me too, my custom qmk layout was working fine one day and the next day it just stopped. I have tried flashing different layouts and even directly flashed the hex from the site. Like a few mentions in this issue thread, even I am getting only a few keys working that too mapped to the wrong characters. My hardware has no issues.

Only the firmware from [kbfirmware.com](https://kbfirmware.com) seems to be working. It works ok but lacks many features.

Please help, I am missing the sweet macros and tap dance features in QMK.

Any help is appreciated.
Thanks!

ubone, 2019-07-30:
https://github.com/qmk/qmk_firmware/tree/master/quantum/tools

yanfali, 2019-07-31:
OK, so I can probably explain this. kbfirmware is over 3 years old, and they haven't kept up with QMK or the changes in the eeprom layout. My guess is that the eeprom layout is just corrupted and causes the keyboard to boot into an invalid mode as soon as it's read. I would just clear your eeprom using the toolbox and move on with life. This is one of the primary reasons we do not recommend kbfirmware.com. There's no way for us to remain compatible with such an old fork.